ZinggJM/GxEPD #4: GxGDEW042T2 library no-longer compiles for ESP8266 or ESP32

G6EJD, 2017-09-12:
Compile error 'min' function not recognised in GxGDEW042T2 .cpp previous version worked OK.

ZinggJM, 2017-09-15:
Please provide more information.

For test, I downloaded the actual master branch to a separate notebook, installed it with the Library Manager Add .Zip Library method, and compiled the GxEPD_SPI_TestExample with GxGDEW042T2.cpp selected for Wemos D1 R2 & mini without any error.

I had seen issues with min and max defined as macros in ESP8266 libraries on other occasions, and needed to change #include sequence to avoid it. See e.g. IoT_SHT31LP_Example_1.54inchEPD.

G6EJD, 2017-09-15:
I have now solved this by using the latest release, so thank you. It was the min/max macros that were causing the problems. 

Have you any experience of the colour modules (1.54, 2.7 and 2.9) all of which require a different set of commands, although I have found your 'GxGDEW042T2.cpp' library operates (without colour support) the 2.9" Waveshare tri-colour screen, I think they tend to use the same controller chip.

closed 2017-09-15 by G6EJD

ZinggJM, 2017-09-15:
The GxGDEW027C44 is part of the library. It is the only color e-paper module I have.
Sponsors are welcome, as I only invest in hardware I am interested for my use or for learning.
Adapting the demo code downloadable from good-display.com for my library is quite straightforward, at least for me.

G6EJD, 2017-09-15:
I have the following Waveshare products:
2.9" color 296x128
2.7" color 296x176
1.54" color 200x200
So I can test out any new library additions for you. I have not tried your 2.7" version yet.