ZinggJM/GxEPD #5: PROGMEM not working on ESP8266 but OK on ESP32

G6EJD, 2017-09-16:
Compiles OK, available RAM reported as 28612bytes, but run-time error (29), works OK on ESP32 platform.

ZinggJM, 2017-09-17:
Please be at least slightly more specific. Which example program with which display class, or attach your program if this fails. And a translation of run-time error (29) would also be helpful.
Sounds like a missing pgm_read_byte() somewhere.

G6EJD, 2017-09-17:
Hi, I know what the problem is now, it is when free RAM available gets below ~27K, you've recommended 21K, but when JSON decoding, I'm finding the minimum is higher at ~27K, there is clearly other usage of RAM in my programme by JSON buffers and the two consume all available RAM.

closed 2017-09-17 by G6EJD

ZinggJM, 2017-09-17:
I think the ~20k I mentioned in the Good Display Arduino Forum post was the ram usage reported by the compiler for ESP8266 and 4.2inch display, without the bitmap examples.
The buffer needed for the 4.2 is 300*400/8 bytes = 15'000 bytes.
Without PROGMEM, one full screen bitmap needed another 15'000 bytes, because the compiler allocated a copy for the const char array, because the flash memory allows only 32bit access.

So, for every b/w e-paper a buffer of x-size * y-size / 8 is needed, except for AVR, where a reduced size buffer is used to fit in ram, usable with the drawPaged method.

The GxEPD library has no control about the ram used by the application.