ZinggJM/GxEPD #7: Missing Min/Max functions for the ESP32

G6EJD, 2017-09-29:
Hi,
This library (GxGDEP015OC1.cpp for the ESP32) needs Min and Max defining:

#define min(a,b) ((a)<(b)?(a):(b));
#define max(a,b) ((a)>(b)?(a):(b));

A couple of people have asked me how to resolve it, your library could usefully be updated.

Regards

Dave

G6EJD, 2017-10-02:
Actually needs to be included in all libraries for the ESP32

ZinggJM, 2017-10-12:
For the package esp8266 min and max are defined in Arduino.h.
Arduino.h is included in GxEPD.h.
All examples compile ok for ESP8266 and ESP32.

The issue is not from library GxEPD, but from some libraries of the ESP8266 package, that #undef min and max.

The solution is to include these mis-behaving libraries after the GxEPD display class.

I had seen this issue with my example IoT_SHT31LP_Example_1.54inchEPD, and needed to arrange the include sequence to solve it.

closed 2017-10-12 by ZinggJM