ZinggJM/GxEPD #9: ESP32 - includes not found in GxEPD.h

jayfn, 2017-10-14:
Tried to switch my working project over from ESP8266 to ESP32.  When switching the libraries are not found anymore in GxEPD.h.

`In file included from C:\...\Arduino\libraries\GxEPD\examples\GxEPD_SPI_TestExample\GxEPD_SPI_TestExample.ino:40:0:

C:\...\Arduino\libraries\GxEPD/GxEPD.h:5:17: fatal error: SPI.h: No such file or directory

compilation terminated.`

Switching back to another board (Arduino, ESP8266) does work again.

Any thoughts?

G6EJD, 2017-10-14:
Check your library address as it looks to me you have a forward/ in the address so maybe a typo. I am regularly switching between 8266 and 32 so there is no issue with the library. Also always make sure you define all EPD stuff then WiFi etc after as some libraries undefine functions that then cause compilation issues.

G6EJD, 2017-10-14:
C:...\Arduino\libraries\GxEPD/GxEPD.h:5:17: fatal error                                        ^

jayfn, 2017-10-14:
Found the error... username contains a german umlaut. thats a no go for the arduino ide in this case! Thanks a lot.

closed 2017-10-14 by jayfn