ZinggJM/GxEPD #10: GxGDEW029Z10 Driver

G6EJD, 2017-10-15:
Thanks for this latest update, I notice the right hand-edge of text and graphics is being cropped short, it looks like about 5-8 pixels are unused, the 2.9B/W version covers the full 296 pixels.

G6EJD, 2017-10-15:
[image]

Some text is missing pixels in a series of vertical line that traverse the whole screen (128 pixels), you can see these lines appear during screen refresh.

ZinggJM, 2017-10-16:
It would be easier for me to analyze your issue, if you provide more details.

Is your issue related to partial update?
What Arduino processor is used, e.g. AVR, ESP8266?

G6EJD, 2017-10-16:
Same with ESP32 or ESP8266. Screen update is full. Same code is OK on BW variant. Only difference is add colour RED for the text.
You can just about make out a section/line on the screen at the far right about 8-10 pixels wide that can't be written to.
There are random bands vertically down the screen and appear across the whole width, but always in the same character position check the letter 'e' and 's' about pixel 5 in the character gird and middle of a capital letter.

ZinggJM, 2017-10-16:
I do not see your issue on my 2.9inch b/w/r display.
No random stripes, no part that shows nothing.

With GxEPD_SPI_TestExample the text starts on the connector side and is complete, no stripes.
With drawCornerTest() uncommented, all four corners show rectangles.

What happens on your display with this test?
Is your BUSY line connected correctly?

You could also uncomment drawCornerTest(); all four corners should show squares.

You could provide a short example that shows your issue to verify on my display.

G6EJD, 2017-10-16:
I'll run the tests now, but this is a black screen on a 2.9" b/w/r display. The reason I thought it was not drawing to the right-hand edge was down to partial erase of the display.
[image]
Busy is connected to Gpio-2 nothings happens if disconnected.

G6EJD, 2017-10-16:
If I modify the library GxGDEW042T2 to:
#define GxGDEW042T2B_WIDTH  128
#define GxGDEW042T2B_HEIGHT 296
The display works OK but no RED obviously, but no banding.
[image]
[image]
So the display hardware is OK, it's a Waveshare 2.9inch e-Paper Module (B) which I just received from Waveshare.

G6EJD, 2017-10-16:
Results from SPI Test:
[image]
[image]
[image]

Is your a Waveshare display, mine may have later firmware.

G6EJD, 2017-10-16:
I've asked Waveshare if they think there is RED buffer (internal) fault) as it's only when the RED layer is added that the lines appear.

ZinggJM, 2017-10-16:
Either you use a modified library, or your display is different.

On my display, the Waveshare logo is not inverted.

And the stripes can't be explained with software.

Your SPI signals might be distorted?

How long does the update take? this should be nearly 15 seconds.

You could uncomment debug output in _waitWhileBusy.

ZinggJM, 2017-10-16:
> Busy is connected to Gpio-2 nothings happens if disconnected.

What do you mean by "nothing happens"?
Is Gpio-2 equal D2, I do not have the mapping at hand.

G6EJD, 2017-10-16:
Gpio-2 is on the ESP32 but I've moved the Busy to Gpio-4 now as Gpio-2 is mused for boot mode too. When I remove the busy signal, the display is not updated, showing busy is connected.

It takes ~16-secs to complete a refresh, I'll try the 'debug' option. It's increasingly looking like a Red buffer (hardware) issue to me. Waiting for a Waveshare response on that.

In the same test bed all other displays are working OK 1.5, 1.5B, 2.13, 2.13B, 2.7, 2.7B, 2.9, 2.9B(? issues) and 4.2 so from that SPI cabling should be OK, connectors at all the same.

ZinggJM, 2017-10-16:
What do you mean by "when the RED layer is added"?

Please note that Dalian Good Display has updated Driver information for this display: controller is IL0373.
Like the controller of the 4.2" it can control b/w or b/w/r. One configuration bit controls b/w or b/w/r.
For b/w the two buffers are used to optimize refresh, for b/w/r for black and red.
This may make taking conclusions out of your experiments difficult.

G6EJD, 2017-10-16:
OK, I meant Red buffer. I've just tried replacing all RED with BLACK and banding problems go, so definitely a RED issue.
I think the display is faulty.

ZinggJM, 2017-10-16:
Have you ever tried to fill the screen fully red, fully black, fully white?

What happens if you draw a full red bitmap directly to the display?

Do you have enough free RAM, or might there be overwrites?

G6EJD, 2017-10-16:
Interest idea, free RAM is 253780 bytes (ESP32). Display test white, black and red:
[image]
[image]
[image]

So issue exists on Black and Red which suggests stuck pixels at those locations (0xFF), white is all 0xFF.

Programme is just fill screen then display update.

ZinggJM, 2017-10-16:
Ok, I would not call that random, rather reproducible.
And even with your test with modified class GxGDEW042T2 you can faintly see the same lines.

This looks like some source or gate lines (too lazy to look up which is for x and which for y) are either stuck or not good connected (the gold bumps of the driver chip). Ask for replacement.

G6EJD, 2017-10-16:
What's interesting is when I use the modified 42T driver it works OK, no lines, no ghosting, I just tried a black fill screen and perfect. But when I refresh the screen the lines persist, so those row/columns are not being cleared/reset and a fault is highly likely. No response from Waveshare yet, I'll let you know what their response is. 

Many thanks for your help and perseverance in getting this far.

ZinggJM, 2017-10-19:
Closed, because this is not an issue of the GxEPD library.

closed 2017-10-19 by ZinggJM

G6EJD, 2017-11-10:
Waveshare technical support say these are driver issues, nothing wrong with the display, they note the display driver chip is different from the 2.9" b/w variant and the driver waveforms are incorrect.

With the latest 29Z driver it displays a black image perfectly then the image is let's call it partially greyed out:
Perfect image just before RED is applied:
[image]

Then after a series of flashing, the final image:
[image]

ZinggJM, 2017-11-11:
If I look at the whole history of this issue, I get the impression, your 2.9" e-paper display shows different effects or defects over time.

Your newest post shows an effect that could best be explained by a defective waveform table.
For the 3-color e-papers and the e-papers that are not advertised as having partial update, the waveform table (LUT) is taken from the controller internal OTP.
The actual demo code for these displays unfortunately do not provide LUT tables, as these are not needed. The one exception is the GDEW027C44 demo, which has the LUT tables, but also uses the values from OTP.

If we had these tables, we could load the tables (to the registers), and see if it makes a difference. And if it did, we could re-program the OTP.

You are the buyer of the display that shows the issue, so you could ask the vendor for the table values.

G6EJD, 2017-11-11:
With the exception of the 2.9"B display all my others are working great from 1.54" up to 4.2". Waveshare keep saying the controller is different, but their not the manufacturer of the driver and despite my number examples of image problems they refuse to accept there's a problem. It's all very strange really. It does work perfectly until the red buffer is sent, the images above were taken about 2-secs apart, even in black it's a good very clear image, then suddenly starts the 'fade-out'. 

I'm still waiting for another response from the Waveshare engineers, but I'm with you none of this can really be explained by the software, suggesting a faulty part. After all it was OK.

Thanks for the support. Waveshare could usefully adopt your drivers as the de-facto driver for their units, as their support (drivers) are almost useless IMO.