ZinggJM/GxEPD #11: powerDown()

G6EJD, 2017-10-15:
I notice that the display continues to draw ~1.5mA when at rest, but the issue of a powerDown() makes it sleep and reduces the power to near 0.

Could you add powerDown to the main functions so that the display can be put to sleep please.

ZinggJM, 2017-10-16:
It would be easier for me to analyze your issue, if you provide more details.

What do you mean with "main functions"?

By design, all methods that do full update end with power off,
all methods that do partial update do not power off, but there is the powerDown() method available.

I do not know what effect on degradation/ghosting and life time power cycling with partial update has.
So I leave the option to power off after partial update to the application.

G6EJD, 2017-10-16:
Not all of your libraries allow access to the powerdown() function e.g. _powerDown() so they can't be called by the application if needed. I thought it would be useful to provide the function outside of the library.

ZinggJM, 2017-10-19:
All classes for SPI e-papers now have a powerDown() method.

closed 2017-10-19 by ZinggJM

habkorn, 2018-08-21:
Hi,
I've noticed that after using the powerDown() function the white areas of the display get a grey-ish. Do you know a method to mitigate this problem? I'm using the 200x200 pixel GxGDEP015OC1.
Thank you for your great and useful library.

chron0, 2018-11-07:
After reading "all methods that do partial update do not power off, but there is the powerDown() method available" I've tried adding display.powerDown() after a partial update within my loop. Power consumption decreases but the next iteration in the loop doesn't update the display anymore. I couldn't find an accessible wakeup call in the sources. Do we have to do a full update to get the display back up or is there another way I'm missing to keep power consumption low during partial updates?

G6EJD, 2018-11-07:
I think this is going to come down to trying to get the display to work outside the law of physics, as the display needs time for the pigment particles to be polarised and arrive at their final destination, perhaps don’t do a power down immediately but after a delay because they are still in transition. The displays are woken by an init / transition of the reset line so you can simulate that. I think the displays were never intended to be active / frequent updates and can’t be used to do so, which is why the ghosting eventually happens, but as we know that’s the physics at play as the pigment particles eventually get polarised and need to be discharged/reset.

I don’t have an Amazon Kiindle but wonder if they can do partial updates as the technology seems to be similar?

chron0, 2018-11-07:
Ah, G6EJD :) I've been looking at your publications about EPDs as well to get me started. Kudos and thanks to both of you @ZinggJM and @G6EJD for your work in the field and sharing it. Prior to https://www.youtube.com/watch?v=MsbiO8EAsGw I didn't look further into waveshare EPDs, since the redraw always looked so crude (full white, black, white) and seemed so long compared to a kindle paperwhite. On the kindle, the page stays white, only the text itself is redrawn. It looks like they're doing some fancy "random" pixel/inkbin redraw, where lettes dissolve and get redrawn, but given the amount of chips in the kindle I wouldn't be surprised if there is some FPGA/ASIC at work doing the "magic". 

My current setup is an esp8266 with a 4.2" w/b waveshare running on battery, updating every 5 minutes during the day and deep sleeping in the night to get maximum battery runtime. With current GxEPD, I have no ghosting issues, not even the border is "flashing" anymore. But since I need to squeeze every mAh out that I can, I'd like to put the display in powerdown. After doing partial updates, go to sleep for 5 minutes and then wake up and do another partial redraw until it goes to deep sleep for the night and resets the next morning with a full redraw.

G6EJD, 2018-11-08:
In terms of the genesis of this issue/topic, I have just measured the power consumption of a 4.2" display in a powerdown() state and it consumes 21uA (using a Fluke 87) which confirms the powerdown function is working correctly.

This small residual current is almost certainly associated with supporting display circuitry, usually a resistive load of some sort, as clearly the display retains its image after complete power removal, so the result is not unusual.

@chron0 This measurement points to your ESP8266 being the dominant power consumer and I think you should run your project with and without a display to determine the ratio between ESP8266 and display power consumption.  Partial updates will reduce power consumption but not significantly is my experience. I also find it informative to go through the mAHr calculations to see if theory and practice broadly align, they should do.
As an example, I get about 6-months from a 18650 2600mAhr LiPo using an ESP8266 at 15-min weather update cycles using WiFi and a 4.2" e-paper display, but I have spent a long time on code optimisation for power reduction and also my power supply. The most efficient set up is to use a LiFe 3.2v cell which is the same size as the 18650 and so can directly power the ESP8266 with no intermediate power wasting regulator.
Hope this helps.