ZinggJM/GxEPD #12: Problem with Arudino example

areseral, 2017-10-18:
Arduino 1.8.5

If I try run example I got this errors:

In file included from C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxGDEH029A1/GxGDEH029A1.cpp:31:0,

                 from C:\Users\arese\Documents\Arduino\libraries\GxEPD-master\examples\GxEPD_SPI_TestExample\GxEPD_SPI_TestExample.ino:47:

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxGDEH029A1/GxGDEH029A1.h: In constructor 'GxGDEH029A1::GxGDEH029A1(GxIO&, uint8_t, uint8_t)':

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxGDEH029A1/GxGDEH029A1.h:119:13: error: 'GxGDEH029A1::_busy' will be initialized after [-Werror=reorder]

     uint8_t _busy;

             ^

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxGDEH029A1/GxGDEH029A1.h:116:13: error:   'int16_t GxGDEH029A1::_current_page' [-Werror=reorder]

     int16_t _current_page;

             ^

In file included from C:\Users\arese\Documents\Arduino\libraries\GxEPD-master\examples\GxEPD_SPI_TestExample\GxEPD_SPI_TestExample.ino:47:0:

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxGDEH029A1/GxGDEH029A1.cpp:65:1: error:   when initialized here [-Werror=reorder]

 GxGDEH029A1::GxGDEH029A1(GxIO& io, uint8_t rst, uint8_t busy) :

 ^

In file included from C:\Users\arese\Documents\Arduino\libraries\GxEPD-master\examples\GxEPD_SPI_TestExample\GxEPD_SPI_TestExample.ino:64:0:

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO_SPI/GxIO_SPI.cpp: In member function 'virtual void GxIO_SPI::writeData16(uint16_t, uint32_t)':

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO_SPI/GxIO_SPI.cpp:219:21: error: narrowing conversion of '(((int)d) >> 8)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Werror=narrowing]

   uint8_t b[2] = {d >> 8 , d};

                     ^

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO_SPI/GxIO_SPI.cpp:219:29: error: narrowing conversion of 'd' from 'uint16_t {aka short unsigned int}' to 'uint8_t {aka unsigned char}' inside { } [-Werror=narrowing]

   uint8_t b[2] = {d >> 8 , d};

                             ^

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO_SPI/GxIO_SPI.cpp: In member function 'virtual void GxIO_SPI_USING_TRANSACTION::writeData16(uint16_t, uint32_t)':

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO_SPI/GxIO_SPI.cpp:635:21: error: narrowing conversion of '(((int)d) >> 8)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Werror=narrowing]

   uint8_t b[2] = {d >> 8 , d};

                     ^

C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO_SPI/GxIO_SPI.cpp:635:29: error: narrowing conversion of 'd' from 'uint16_t {aka short unsigned int}' to 'uint8_t {aka unsigned char}' inside { } [-Werror=narrowing]

   uint8_t b[2] = {d >> 8 , d};

                             ^

cc1plus.exe: some warnings being treated as errors

ZinggJM, 2017-10-19:
For which "Arudino" do you compile? board, package?
What level of compiler warnings is selected in your preferences?

With a fresh installed Arduino 1.8.5 the warning level is "none".
When set to "More" or "All" I get these warnings, but not as error.
With "Default" I get no warnings.

Thank you for reporting this issue, I may consider to modify to solve some of the warnings for a future version, but I consider this not as important.

areseral, 2017-10-19:
Esp32
I check level later.

I have neweest version because I hopes that repair problem. But no, error still has.

ZinggJM, 2017-10-19:
I confirm there is an issue with warnings from the compiler of the package ESP32.
I had assumed it to be the same compiler as for ESP8266, but it is more strict.
But it compiles anyway with warning level "Default" or lower.

I had not consulted issues before my latest commit, so I did not know this issue in time.
As there is a workaround (lower warning level), I will only fix it in other updates to the display classes.

G6EJD, 2017-10-19:
I confirm using Default as the compiler warning solves the issues on the ESP32. The 16-bit data type that is being converted to 8-bit is causing the issue and seems easy to fix by changing to 8-bit, that's what I have done to prove the error can be fixed in my own copy of the libraries. Thanks for the library, it is very good and useful, actually I think it is the only working library for the e-Paper displays currently on the market.

areseral, 2017-10-19:
Easy to fix but still change in original library...

Now i have another fail:
C:\Users\arese\Documents\Arduino\libraries\GxEPD-master/GxGDEH029A1/GxGDEH029A1.cpp:281:45: error: 'min' was not declared in this scope

   uint16_t xe = min(GxGDEH029A1_WIDTH, x + w) - 1;

                                             ^

ZinggJM, 2017-10-19:
This would be an other issue, but this has been answered in issue #7.

As I already answered, THIS issue has a workaround and will be fixed by and by.

So please let me work and don't bother me with issues that have been answered.

ZinggJM, 2017-10-21:
Relevant warnings are fixed.

closed 2017-10-21 by ZinggJM