ZinggJM/GxEPD #14: [Request] Add Waveshare 2.7 b/w

ccvh, 2017-11-13:
Hey, 

i have the following e Paper from Waveshare: https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT
The 3-Color-Version Library doesn't seem to work unfortunately. Is there a chance you'll add this display?

Thanks in advance! Greetings

ZinggJM, 2017-11-13:
There would be a chance. But I do not have the 2.7" b/w e-paper, and have no personal use for it.
To be able to support it, I would need one permanently. Therefore any sponsor would be welcome.

G6EJD, 2017-11-13:
I've suggested to Waveshare they should sponsor you, because it is people like you who make their displays more accessible. I've also asked for the 2.9"B datasheet, but they seem reluctant to do that, or more likely they don't have access to it themselves - as you know they don't manufacture the displays.

BTW I notice they have bumped up your comments about providing drivers on the AliExpress store, so yours is the first a customer sees.

ZinggJM, 2017-11-27:
Support for this display will be added. I ordered this display, because I am interested in experimenting with waveform tables for partial update.

closed 2017-11-27 by ZinggJM

ZinggJM, 2017-12-22:
Support for this display is added. No optimized waveform table for partial update yet.

pskowronek, 2018-09-09:
Any hope for partial refresh?

G6EJD, 2018-09-09:
Extract from the data sheet for 2.7” display:
‘This module doesn’t support partial refresh’

pskowronek, 2018-09-09:
Yes I know, but I was hoping for optimized waveform, like this guy did: http://benkrasnow.blogspot.com/2017/10/fast-partial-refresh-on-42-e-paper.html

G6EJD, 2018-09-09:
Yes but he’s using a 4.2” that does support (display driver chip) partial refresh.

pskowronek, 2018-09-09:
I thought he modified LUT tables to shorten amount of passes over each "pixel" so it is now quicker (and of course may leave some left overs) and that wasn't related to "partial" refresh per se.
When I referred to "partial refresh" I was rather thinking about "faster" refresh by using "faster" LUT tables for waveforms.

pskowronek, 2018-09-18:
The faster LUT tables for 2.7" work too. When using waveshare lib epd2in7.py modify lut_vcom_dc, lut_ww, lut_bw, lut_bb, lut_wb arrays and set last value of each row to 0x01 except for the last 3 rows. This lowers refresh time to about 1s for black and around 3s for red. Of course left overs may build up, so from time to time one would need to do proper refresh. Disclaimer - these are my experimental values, test it on your own responsibility :)