ZinggJM/GxEPD #16: Support for Chinese Font

evan-opener, 2017-11-22:
Hi,
Thanks for the library. I use this for my waveshare 7.5" b/w with wemos D1 mini. It works well.

I just want to know if it can support Chinese font.

I tried to use the Adafruit GFX font convert tool but the .h file still has no Chinese Font. Is there any method to do that?

G6EJD, 2017-11-22:
According to Waveshare English and Chinese fonts are in-built and I see Chinese characters sometimes, especially for ASCII numbers above 128, I thought there was a specific switch to enable Chinese fonts, like for English I use ASCII_32GB and I thought Chinese was ASCII_32GBK (32,48 and 64 sizes).

No one seems to know what the font format is, Waveshare told me the fonts were produced on an WinXP system using an old tool that is no-longer available!

evan-opener, 2017-11-22:
@G6EJD 
Thanks for you reply.
Actually the origin demo I got from Waveshare for Arduino was very simple. They have more detail explaination on RaspberryPi but not too much on Arduino. Only the basic rendering for one page frame in arrays or bitmaps. That's why I use the GxEPD library.

ZinggJM, 2017-11-27:
Adding fonts to a library that uses Adafruit_GFX or to an application that uses such a library is of general interest. So maybe you could ask your question in the Arduino Forum Displays. There are many experts willing to help.
I have no experience in adding fonts yet, nor with the font convert tool.
Does your issue relate to the result of the font convert tool, or the access to the generates font files?

ZinggJM, 2018-01-31:
Adding fonts to Adafruit_GFX or converting fonts to free font format is discussed in the Arduino Forum Displays. Adding fonts to GxEPD is outside of my current work and capacity.

closed 2018-01-31 by ZinggJM