ZinggJM/GxEPD #17: Problem with min() and max()

gztproject, 2017-11-30:
Compiler spat out the error for each min() and max() function in the library:
`'std::min'      min(const _Tp& __a, const _Tp& __b, _Compare __comp)`

After some googling I found [this](https://github.com/esp8266/Arduino/issues/263) and replaced all min() and max() with _min() and _max() as suggested in that issue and it compiled.

G6EJD, 2017-11-30:
You can also solve this problem by moving the EPD includes higher up the source code so they are the first to be seen by the compiler.

sidoh, 2017-12-10:
Even `#include <vector>` triggers this.  I'm not sure where `_min` and `_max` are defined, but using these instead works great for me as well.

G6EJD, 2017-12-10:
You can prevent this problem by moving the GxEPD includes to be the first as other libraries can undefine min and max creating the problem

sidoh, 2017-12-10:
Sorry, I should've mentioned that I tried that, and it also worked.

I meant to point out that even stuff in `std` like `vector` will trigger this, and that it seems like a hard sell to say that `vector` is misbehaving.  

I think it'd be nice if GxEPD were robust to stuff like `vector` getting included before it, and it seems like switching to `_min` and `_max` does the trick.  Maybe there are platform concerns (I've only tested on ESP8266 and ESP32), but would think that's a pretty easy obstacle to overcome :)

sidoh, 2017-12-10:
In case it's helpful, my fork has `min` and `max` replaced with `_min` and `_max`:

https://github.com/sidoh/GxEPD/commit/5c075bc25d0b4b1e3cfe4712d49ff603440d8adf

G6EJD, 2017-12-10:
That’s the solution so this can’t happen but why do some libraries undefined the functions really they are at fault

sidoh, 2017-12-10:
What I'm trying to say is that whatever is going on with`min` and `max` appears to be triggered in stdlibc++, so it's not like it's just a dinky library doing something dopey.  We're talking about _the_ core language library that is linked to from ESP8266/ESP32 Arduino.

It looks like it's well known that `std::vector` in particular is incompatible with Arduino ([reference](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/tests/host/common/Arduino.h#L249-L253)).

My guess is that all of the libraries that cause issues with `min` and `max` are undefing them so that they can use `std::vector`.  Looks like the very popular [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/blob/8139925eb9b5d0d159d36edc03a409788138a627/src/WebResponseImpl.h#L24-L29) is one of them.

What makes sense to me either:

* Using `_min` and `_max` instead.  For what it's worth, they're defined in [Arduino.h](https://github.com/esp8266/Arduino/blob/2.4.0-rc2/cores/esp8266/Arduino.h#L256-L257) in the ESP8266 SDK.
* Defining custom macros for min and max and using those instead.  This seems like the safest option, given that it's a very simple operation :)

ZinggJM, 2017-12-17:
So I think it is time for me to address this issue.

I used min() and max(), because they are available in the Arduino IDE, available with any #include of Arduino.h, and because these names best describe what they do, and are completely type independent.

The root of the issue comes from the fact that I need to include .cpp files in the application, because of the directory structure I use in my library, which is not directly supported by the Arduino IDE, the way it decides which .cpp files are automatically included in the build process for libraries.

With all the explanations posted in this issues, especially by sidoh, thanks, I have decided I will replace min() and max() in my library. But as all e-paper device classes are affected, this will not be immediate.

Thanks to all of you.

ZinggJM, 2017-12-18:
min() and max() are replaced.

closed 2017-12-18 by ZinggJM

sidoh, 2017-12-18:
Awesome.  Thank you @ZinggJM!