ZinggJM/GxEPD #19: busy timeout

pmimsecu, 2018-01-08:
I am using your library to run a 4.2 epaper display from an ESP32.  I can only update the display once then I get a Bust Timeout! response in my serial monitor.  Here is the code.  The loop begins and it works only once.

`

/************************************************************************************
   GxEPD_SPI_TestExample : test example for e-Paper displays from Dalian Good Display Co., Ltd.: www.good-display.com

   based on Demo Example from Good Display, now available on http://www.good-display.com/download_list/downloadcategoryid=34&isMode=false.html

   Author : J-M Zingg

   Version : 2.0
  
/* ESP32 SPI WIRING SETUP
 * RED    =   3.3V
 * BLACK  =   GND
 * BLUE   =   DIN   =  MOSI  =   D23   
 * YELLOW =   CLK   =  SCK   =   D18     
 * ORANGE =   CS    =  SS    =   D5
 * GREEN  =   DC    =  DC    =   D22
 * WHITE  =   RST   =  RESET =   D21
 * PURPLE =   BUSY  =  MISO  =   D19
 * 
 */

// include library, include base class, make path known
#include <GxEPD.h>

#include <GxGDEW042T2/GxGDEW042T2.cpp>      // 4.2" b/w epaper

// FreeFonts from Adafruit_GFX
#include <Fonts/FreeMonoBold9pt7b.h>
#include <Fonts/FreeMonoBold12pt7b.h>
#include <Fonts/FreeMonoBold18pt7b.h>
#include <Fonts/FreeMonoBold24pt7b.h>

#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>

#if defined(ESP8266)

// GxIO_SPI(SPIClass& spi, int8_t cs, int8_t dc, int8_t rst = -1, int8_t bl = -1);
GxIO_Class io(SPI, SS, 0, 2); // arbitrary selection of D3(=0), D4(=2), selected for default of GxEPD_Class
// GxGDEP015OC1(GxIO& io, uint8_t rst = 2, uint8_t busy = 4);
GxEPD_Class display(io); // default selection of D4(=2), D2(=4)

#elif defined(ESP32)

// pins_arduino.h, e.g. LOLIN32
//static const uint8_t SS    = 5;
//static const uint8_t MOSI  = 23;
//static const uint8_t MISO  = 19;
//static const uint8_t MOSI  = 8;
//static const uint8_t MISO  = 10;
//static const uint8_t SCK   = 18;
/*
// GxIO_SPI(SPIClass& spi, int8_t cs, int8_t dc, int8_t rst = -1, int8_t bl = -1);
//GxIO_Class io(SPI, SS, 17, 16); // arbitrary selection of 17, 16
GxIO_Class io(SPI, 17, 16, 5); // arbitrary selection of 17, 16
// GxGDEP015OC1(GxIO& io, uint8_t rst = D4, uint8_t busy = D2);
GxEPD_Class display(io, 5, 19); // arbitrary selection of (16), 4
//Serial.println("board: ESP32");
*/
//Serial.println("ESP32 board");
//GxIO_SPI(SPIClass& spi, int8_t cs, int8_t dc, int8_t rst = -1, int8_t bl = -1);
GxIO_Class io(SPI, 5, 22, 21); // arbitrary selection of 17, 16
//GxIO_Class io(SPI, 5, 19, 4); // arbitrary selection of 17, 16
//GxIO_Class io(SPI, 17, 16, 5); // arbitrary selection of 17, 16

//GxGDEP015OC1(GxIO& io, uint8_t rst = D4, uint8_t busy = D2);
//GxEPD_Class display(io, 4, 2); // arbitrary selection of (16), 4
GxEPD_Class display(io, 21, 19); // arbitrary selection of (16), 4
//Serial.println("board: ESP32");

#elif defined(ARDUINO_ARCH_SAMD)

Serial.println("Arch SAMD");

GxIO_Class io(SPI, 4, 7, 6);
GxEPD_Class display(io, 6, 5);

#elif defined(_BOARD_GENERIC_STM32F103C_H_)

// GxIO_SPI(SPIClass& spi, int8_t cs, int8_t dc, int8_t rst = -1, int8_t bl = -1);
GxIO_Class io(SPI, SS, 8, 9);
// GxGDEP015OC1(GxIO& io, uint8_t rst = 9, uint8_t busy = 7);
GxEPD_Class display(io, 9, 3);

#else
GxIO_Class io(SPI, SS, 8, 9); // arbitrary selection of 8, 9 selected for default of GxEPD_Class
//GxIO_DESTM32L io;
//GxIO_GreenSTM32F103V io;
GxEPD_Class display(io);

#endif

/////////////////////////////////////////////////////////SETUP 
#include "WiFi.h"

char ssid[] = "ssid";    
char password[] = "password";

int counter;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("setup");
  Serial.println("sketch_jan07b.ino");

  WiFi.begin(ssid, password);
  delay(500);
  Serial.println("connecting to " + String(ssid) + ":" + String(password));
    while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
   Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    delay(1000); 
 WiFi.disconnect();
delay(1000);

 Serial.println("disconnected");
 display.init();
    
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  //display.setRotation(45);
  display.setCursor(0, 30);
  display.setFont(&FreeMonoBold24pt7b);
  display.setCursor(50, 110);
  display.println("Connected");
  display.update();
  delay(500);
  Serial.println("setup done");
 }

void loop()
{
screen();
}

void screen(){
  counter++;
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(&FreeMonoBold24pt7b);
  //display.setRotation(45);
  display.setCursor(150, 150);
  Serial.println("before update");
  display.println("LOOPING");
  display.println(counter);
  display.update();
  Serial.println("after update");
  delay(500);
  Serial.print("Loops: ");
  Serial.println(counter);
  
  delay(5000);
  
}

`

G6EJD, 2018-01-08:
Your loading a lot of fonts, I wonder how much free ram you have left, I would try it by commenting out the unused fonts.

pmimsecu, 2018-01-08:
Tried removing all but one font and it didnt help.  I checked the source file and it issues a busy timeout when the busy pin (19) is high.  Im wondering if I have a wiring issue.

Here is the section that issues the message (I know you don't need me telling you about the library since you wrote it, thanks for that btw)

`

void GxGDEW042T2::_waitWhileBusy(const char* comment)
{
  unsigned long start = micros();
  while (1)
  { //=0 BUSY
    if (digitalRead(_busy) == 1) break;
    delay(1);
    if (micros() - start > GxGDEW042T2_BUSY_TIMEOUT)
    {
      Serial.println("Busy Timeout!");
      break;
    }
  }
  if (comment)
  {
#if !defined(DISABLE_DIAGNOSTIC_OUTPUT)
    unsigned long elapsed = micros() - start;
    Serial.print(comment);
    Serial.print(" : ");
    Serial.println(elapsed);
#endif
  }
  (void) start;
}

`

G6EJD, 2018-01-08:
I'm using this library a lot (ESP32) and don't get any issues. Are your SPI cables long, say greater than 10cms, if so you may be getting transmissions errors. I checked you code and pin designations looked OK. What I did not do is cross-check for pins that are already defined for other uses, as I've found there are sometimes conflicts.  SPI is predefined for most boars to 23,18 and 5 and your using 21, 22 which are the I2C bus, it may be worth a trial moving them to other pins, then I spotted these:
GxIO_Class io(SPI, 17, 16, 5); // arbitrary selection of 17, 16
GxIO_Class io(SPI, 5, 22, 21); // arbitrary selection of 17, 16

So who knows what pins are being used :) almost certainly the last definition, so got to be worth removing that duplicate to remove any possibility, I don't know how the compiler operates.

pmimsecu, 2018-01-08:
First of all, many thanks for sharing this amazing library with us.  Its great to finally be able to use epaper.  Thank you for what you do.

I tried other pins and it does the same thing.  It might be an issue with WiFi since it also uses the I2C.  This is why I run WiFi.disconnect() before updating the display.

I was finally able to make it work by running display.init(); at the beginning of the loop then running display.powerDown(); at the end of the loop.  My ESP32 board is diffierent than others for some reason.  the bottom right pin closest to the BOOT button is CLK while most other ESP32 boards Ive seen has 3.3V as the first pin. (https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png)

I guess I would consider this partially solved because I was able to make it work, but I still think there is something with my board thats not right.

ZinggJM, 2018-01-13:
Please tell us if you think it has something to do with the library. Close the issue otherwise.

pmimsecu, 2018-01-13:
I never really solved it but I will close the issue because I was able to work around it.  I think it has something to do with my breakout version.  The pinout is not the same as other ESP32 boards I have seen.

closed 2018-01-13 by pmimsecu