ZinggJM/GxEPD #21: About the include path problem of newest Arduino IDE

alieonsido, 2018-01-15:
Hello,
my work environment is Windows 10 and I tried to test the EPD & Example, 
but I found that when including, the relative path didn't work.

like this:
`GxEPD_SPI_TestExample.ino:47:19: fatal error: GxEPD.h: No such file or directory`
my IDE is 1.8.1. Also, I Googled to find this:
[the Loading local libraries problem](https://arduino.stackexchange.com/questions/8651/loading-local-libraries)

So, I guess the including path is all mess up for the update of Arduino IDE.

I'm not sure if I'm right, but one thing is true: I can't compiler origin example file.
my compromise is include in the same dir, but it's dirty.
maybe I need a suggestion, or the example/library.

G6EJD, 2018-01-16:
What I do is download the library from Github, unzip and then copy the library file to the Arduino IDE Library folder. Alternatively you can define a specific path to the library with #define RELATIVE_PATH(library) ABSOLUTE_PATH(PROJECT_ROOT, library)

ZinggJM, 2018-01-18:
GxEPD works with the standard method to add a library to the Arduino IDE:
1. download as ZIP file
2. in menu Sketch: Include Library > Add .ZIP Library...

closed 2018-01-18 by ZinggJM