ZinggJM/GxEPD #22: Manually defining pins

mattncsu, 2018-01-18:
Hi, I'm retrofitting the GxEPD library to an existing project based on a ESP8266 Huzzah Feather/Waveshare2.9 EPD and need to define the pins as follows:
 Connect the following pins: 
```
 Display  Huzzah
 BUSY     16
 RST      3
 DC       15
 CS       12
 CLK      14
 DIN      13
 GND      GND
 3.3V     GND
```
Using the comments in the example code, I made my best guess as:
```
#if defined(ESP8266)
  // GxIO_SPI(SPIClass& spi, int8_t cs, int8_t dc, int8_t rst = -1, int8_t bl = -1);
  GxIO_Class io(SPI, 12, 15, 3, 16); 
  // GxGDEP015OC1(GxIO& io, uint8_t rst = 2, uint8_t busy = 4);
  GxEPD_Class display(io, 3, 16); // default selection of D4(=2), D2(=4)
```

However, the screen does not update.  Perhaps, it is because I defined pin 12 as CS which is normally the MISO line.  I will check everything tomorrow with my logic analyzer, but wanted to post early just to see if I even defined the pins properly.

G6EJD, 2018-01-18:
You don’t need ,16 on the IO class line. The rest is OK. But, afaik some SPI pins can’t be redefined and this may be giving you some issues, I have not checked the huzzah pin outs to know.

ZinggJM, 2018-01-18:
For ESP8266 and ESP32 the SS pin is reserved for SS/CS if SPI is used.
You can use a different pin for CS, but you can't use the SS pin for something other.
In your case DC will not work, when assigned to 15, which is usually SS (I did not check for HUZZAH).

closed 2018-01-18 by ZinggJM

mattncsu, 2018-01-19:
I attached the pinout below. I had the pins in the above table working with the [MiniGrafx library](https://github.com/squix78/minigrafx).  Here is what I am working with:
[image]
I have pins 0 and 2 open and both of them have functions realted to boot mode:

> GPIO # 0, which does not have an internal pullup, and is also connected a red LED. This pin is used by the ESP8266 to determine when to boot into the bootloader. If the pin is held low during power-up it will start bootloading! That said, you can always use it as an output, and blink the red LED.
> 
> GPIO # 2, is also used to detect boot-mode. It also is connected to the blue LED that is near the WiFi antenna. It has a pullup resistor connected to it, and you can use it as any output (like #0) and blink the blue LED.

> Also note that GPIO # 12/13/14 are the same as the SCK/MOSI/MISO 'SPI' pins!

[image]

Here what I'm seeing with 16 in the IO line, when I removed it, I only get a flatline.:
[image]

G6EJD, 2018-01-19:
Io2 is spare. 16 is held low by the UART as it’s the device programming port. Suggest you change 16 to 2

G6EJD, 2018-01-19:
You need to match epd inputs to an 8266 must be an output pin if that makes sense, so that during boot the epd will not hold io0 or io2 low during boot. After boot the pins can be used ok.

ZinggJM, 2018-01-19:
I do not consider this as an issue with the library. The Arduino Forum Displays is the place for help with usage.
Or make a suggestion here if "documentation" should be improved, but this is not my strength.

G6EJD, 2018-01-19:
Huzzah    		EPD	I2C
5 (SCL)			SCL
4 (SDA)			SDA
14 (SCK)		SCK
12 (MISO)	DIN
15 (RTS0)		RST
00 (CS2)		CS (EPD input)
02 (DC)		DC (EPD input)
03 (BUSY)		BUSY (EPD output)

ZinggJM, 2018-01-19:
Use MOSI for DIN (Master Out Slave In), EPD is slave.

G6EJD, 2018-01-19:
Thanks, I corrected the error above. I find the Huzzah diagram somewhat confusing with 3 numbers for each pin.

mattncsu, 2018-01-19:
Thanks for the help, I will try that pin arrangement tonight. I agree that I don't think it is a code issue; my original question was more documentation related (interpreting the code comment on pin definition). I've always felt that github could use a discussion section or maybe being able to tag "issues" as questions.

mattncsu, 2018-01-20:
Looks like it is working using:
Huzzah EPD I2C
5 (SCL) SCL
4 (SDA) SDA
14 (SCK) SCK
13 (MOSI) DIN
15 (RTS0) RST
00 (CS2) CS (EPD input)
02 (DC) DC (EPD input)
16 (BUSY) BUSY (EPD output) (swapped from 3 to avoid interfering with uploading sketches)

[image]