ZinggJM/GxEPD #23: Rotate Display

reibuehl, 2018-01-20:
I have a Waveshare 2.9" e-paper display with 296x128 pixels that I would like to use in landscape mode. Regular display.println() writes to the display in portrait mode. I tried display.setRotation() as described in the Adafruit_GFX documentation but all other values than 0 seem to freeze my sketch.
Is there a way to use the display in landscape mode?

G6EJD, 2018-01-20:
I'm not sure what's going wrong, but I'm currently using both the 2.9" bw/ and bw/r displays from Waveshare and they rotate OK. My setup copied from a working sketch is:
  display.init();
  display.fillScreen(GxEPD_WHITE);
  display.setRotation(1); 

How old are they?

Did you buy them from Waveshare?

reibuehl, 2018-01-20:
I bought the display two weeks ago from a German distributor, https://eckstein-shop.de/. It is a b/w/r model. 

I use 
```
#include <GxEPD.h>
#include <GxGDEW029Z10/GxGDEW029Z10.cpp>    // 2.9" b/w/r
#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>
#include <Fonts/FreeSansBold24pt7b.h>
GxIO_Class io(SPI, SS, 0, 2); // arbitrary selection of D3(=0), D4(=2), selected for default of GxEPD_Class
GxEPD_Class display(io); // default selection of D4(=2), D2(=4)

void setup() {
  ....
  display.init();
  ....
}

void loop() {
  const char* name = "FreeSansBold24pt7b";
  const GFXfont* f = &FreeSansBold24pt7b;
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(f);
  uint8_t rotation = display.getRotation();  
  display.setRotation(1);
  display.setCursor(0, 0);
  display.println();
  display.println(rotation);
  display.println("Test");
  display.updateWindow(0, 0, display.width(), display.height());
  delay(10000)
}
```

If I use display.setRotation(1); the display shows the 0 and the text but in the wrong orientation. For every other value (1,2,3), the code seems to freeze without any output on the serial console and before the display gets updated the first time

G6EJD, 2018-01-20:
I see your using partial update.  Have you confirmed the display rotates, perhaps with this:
void loop(){
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(f);
  display.setRotation(1);
  display.setCursor(0, 0);
  display.println("Test");
  display.update();
  delay(5000);
  display.setRotation(2);
  display.setCursor(0, 0);
  display.println("Test");
  display.update();
}

G6EJD, 2018-01-20:
I get a large 0 and Test landscape then if I change display.setRotation(0); portrait using your code example.

reibuehl, 2018-01-20:
Yes, with display.update() instead of display.updateWindow(0, 0, display.width(), display.height()) it works as expected. I wanted to use the partial update later since I have a large fixed part and only a smaller variable part that gets updated more frequently. Maybe I will use the normal update() then.

G6EJD, 2018-01-20:
The partialupdate is as quoted by the author as being experimental, although the test example works OK, I've just run it. I have not tried rotating the display or studied what the true or false parameter does on the display.updateWindow(0, 0, display.width(), display.height(),true); function

reibuehl, 2018-01-20:
I will see that I can get my stuff done without the partial update then. Many thanks for your help!

closed 2018-01-20 by reibuehl

G6EJD, 2018-01-20:
One day Waveshare might issue sufficient documentation to enable partial update to be fully implemented as like your situation it would be very useful to only update a small area of the screen than the whole screen. Their example shows it working, but how?