ZinggJM/GxEPD #24: Simple Text - 2,9 bwr....waveshare

njordan77, 2018-01-22:
Hi, i reduced the hd code to the simplest example for an esp8266+2,9 waveshare bwr display. tried to get some text sent to the display. it works...but strange to me is that initially it is inverted to what i selected (background vs. textcolor)....then a few times the whole screen blinks black white....and in the end my text is shown correctly white background and black text..... any idea what I have done wrong here?

`// mapping suggestion from Waveshare SPI e-Paper to Wemos D1 mini
// BUSY -> D2, RST -> D4, DC -> D3, CS -> D8, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V
#include <GxEPD.h>
#include <GxGDEW029Z10/GxGDEW029Z10.cpp>    // 2.9" b/w/r

#include <Fonts/FreeMonoBold9pt7b.h>
#include <Fonts/FreeMonoBold12pt7b.h>
#include <Fonts/FreeMonoBold18pt7b.h>
#include <Fonts/FreeMonoBold24pt7b.h>

#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>

GxIO_Class io(SPI, SS, 0, 2); // arbitrary selection of D3(=0), D4(=2), selected for default of GxEPD_Class
GxEPD_Class display(io); // default selection of D4(=2), D2(=4)

void setup()
{
  display.init();
}

void loop()
{
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(&FreeMonoBold9pt7b);
  display.setCursor(0, 0);
  display.println();
  display.println("njordan TEST");
  display.println(" !\"#$%&'()*+,-./");
  display.update();
  delay (3000);
}`

G6EJD, 2018-01-22:
Nothing! - your code is OK, it's the way e-Paper displays operate. What your seeing is normal, in simple terms, its clears the display all black, then refreshes white except where your text is. Some of the b/w/r displays can take up to 12-secs to paint the image and flash many times :)

G6EJD, 2018-01-22:
http://benkrasnow.blogspot.co.uk/2017/10/fast-partial-refresh-on-42-e-paper.html

njordan77, 2018-01-22:
hm, thanks a lot for the fast reply....little disappointed by the hardware (even if its the way how this display works)....as i have a 4,3inch BW display from waveshare and with different library (no offence!!!!) it works much faster with no blinking in between....but i guess its different hardware without RED.

its little problem as i have a home information system with smart meter power,....solar power coming in, and at least everey 5mins it should update....so if it takes almost 10secs to update its little annoying...of irritating for some people in the house :-)

G6EJD, 2018-01-22:
Not my library ZinggJM is the library author and afaik his library is the only working and reliable library around. The 4.2" b/w displays work very quickly (about 4-secs) the 2.9"b/w is fastest at 2-secs as is the 1.5"b/w which take almost a single refresh, all of my colour displays (b/w/r) take a long time to refresh, some (2.7") up to 18-secs. The 4.3" UART versions also take a while to refresh.

On some screens (not 4.2) you can do partial refresh of a small screen section, but according to the manufacturer you need to do a full refresh occasionally to stop the screen from becoming polarised. 

They are not really meant for fast changing / active displays, reading OK, a refresh every 5-minutes of so is OK, but that's about all. Your better off with TFT but of course won't get the low power consumption advantages.

I use the 4.3" display a lot and whilst not asleep it works OK, but awake takes ~40-75mA, sometime more. I find the on-board SD card is OK, but limited to 7.3 file names. The on-board NAND memory (you can transfer data/images to it) does not last indefinitely and needs to be refreshed. It's unreliable all-round IMO.  It can only display the 3 embedded 7-bit fonts, there is no method of creating new fonts. There are no special characters e.g. a degree symbol for example, well there is but it's very clumsy to use, but that's all. Also on the 4.3" because they implemented negative logic for WAKE (e.g. a negative pulse to wake - they should have called it WAKE BAR) then for low power work, it wakes the display up when you place an ESP to sleep, this happens because the ESP releases the pull-ups and creates a negative pulse and the screen awakes. I have managed to solve this to about 1 in 30 failed sleep cycles (the display never enters sleep) by placing a 2k2 pull-up resistor on the WAKE line.
Hope this helps.

njordan77, 2018-01-23:
wow, thanks a lot for these insights. highly appreciated.
So, the 4,3inch display is not the best way to implement things....better to have a 4,2inch my interpretation.
Guess this is the reason why 4,3 is not so much found anyway in online stores these days....

One more question regarding 2,9bwr....how does the partial update work....i have the situation that some values only need to be updated ever 30mins - others would be best to update every 5mins.....so its just a small area (several small areas) that need to receive new information....

Does partial update do exactly this? will the update also work faster if only part of the display needs an update, or there is no big difference in terms of updating the information....

G6EJD, 2018-01-23:
Yes the 4.2" is IMO adequately fast enough in b/w, 1.5 to 2 secs is not that slow for a zero power screen.
With partial updates typically a clock application, you can update just the hours/min/secs and in that that lets say smaller section of the screen, it is almost seamless. The library has a partial update example, might be worth a look and Waveshre have done a YT video showing it in action for one of their displays. I think the counting clock like display is another of their examples in their code.

Take a look at drivers from:  https://ugfx.io/ 

Also partial update demo here: https://www.youtube.com/watch?v=eKq5RabWmN0 

The problem is very little documentation is released on partial updates.

G6EJD, 2018-01-23:
Coincidentally I have three development systems running at the moment and at 10:00 UK time I watched them refresh on the hour (they are all running the same code and waking from sleep) the 1.54" and 4.2 screens updated in the blink of an eye, by my measurement a slow blink of your eyes but the 2.9b/w/r display took a good 8-secs to do the same.

njordan77, 2018-01-23:
Thanks a lot for this live review! franky speaking, i did not check too much about the refresh times - and from a table i've seen now from waveshare its exactly what you say with your observation.....4.2 BWR is 18secs!!! the same in BW is 2secs.....

I thought Red would be cool. i have a 4,2BWR as well, will see if it fits my use with partial update....lets hope

njordan77, 2018-01-27:
Hi, 2 short questions that came up playing arround with 4,2inch 3color display now.

Is the ° character already implemented and somehow able to be set...simply print with the symbol does not work, is ignored. 
Second question....is there a way to make fonts bigger than 24pt possible?

Cheers Norbert

ZinggJM, 2018-01-31:
GxEPD supports only the fonts available with Adafruit_GFX.
Adding fonts converted to free fonts format is possible, search the Arduino Forum Displays.
Maybe I will have a look at this later, but this is outside of current work and capacity for GxEPD.

closed 2018-01-31 by ZinggJM