ZinggJM/GxEPD #25: Issues with min()/max() when including <ESP8266WiFi.h>

reibuehl, 2018-01-25:
When I include <ESP8266WiFi.h> into an otherwise working sketch, I get errors "error: 'max' was not declared in this scope" from - in my case - GxGDEW029Z10.cpp from the GxEPD library.
Based on information from https://github.com/esp8266/Arduino/issues/263 and https://github.com/esp8266/Arduino/issues/2073 I went into GxGDEW029Z10.cpp and changed every occurence of min() and max() to _min() and _max() and this fixed the compilation error. Is this something that could be done in GxEPD directly?

G6EJD, 2018-01-25:
It's already been fixed, do you have the latest GxEPD library installed? Also moving GxEPD higher up the include list could have solved the issue.

reibuehl, 2018-01-25:
Apparently my version is too old. Sorry!

closed 2018-01-25 by reibuehl

G6EJD, 2018-01-25:
It is difficult to keep up to date with library changes unless they are registered for the Arduino IDE in which case that does it for you. Good to hear you have it resolved.

reibuehl, 2018-01-25:
Do you have any plans to have this library registered for the Arduino IDE?