ZinggJM/GxEPD #26: Partial update with the GxGDEW027C44

AndreR, 2018-01-31:
In your example, you say that partial update is not fully supported for the GxGDEW027C44. I've tried it and parts of the screen seem to stay static, but the partial refreshing takes around 15–16 seconds. During that, not only the updated area, but also the whole edge of the screen flicker.

Is this a hardware limitation of the three-color display. or is it just not fully implemented yet?

G6EJD, 2018-01-31:
I understand only the 1.54, 2.13 and 2.9 inch screens have driver chips that support partial updates. There is a table in the description of this product that lists various capabilities, the 2.7" is N/A for partial updates, but may work sub-optimally is my reading of this: https://www.aliexpress.com/item/Universal-e-Paper-Raw-Panel-Driver-HAT-Used-to-drive-various-SPI-interface-e-Paper-from/32833050422.html

AndreR, 2018-01-31:
That makes sense. Are there any tricks to hack/speed up the process?

G6EJD, 2018-01-31:
Not really, the physics of the screen operation (how fast the coloured particles move) and driver chip dictate speed. On the displays I've hacked they have failed (column drivers fail or previous image shadows), now I stick to the defaults.

AndreR, 2018-01-31:
Ok :)

ZinggJM, 2018-01-31:
I gave an answer to your same question on the Arduino Forum Displays.
I prefer the forum for questions; issues should be reserved for issues with the library.

I will continue experimenting with waveform tables and refresh speed up, but this takes time, and physics limit speed for 3-color e-paper displays.

closed 2018-01-31 by ZinggJM