ZinggJM/GxEPD #27: please post only issues that are real issues with this library!

ZinggJM, 2018-01-31:
The preferred location for questions and suggestions for enhancement is the Arduino Forum Displays.

[https://forum.arduino.cc/index.php?topic=487007.330](url)

I regularly look at the forum topic, but I look only sporadic if there are issues reported here.

per1234, 2018-02-04:
If you want to make this message more visible you might consider adding it to a CONTRIBUTING.md. See:
https://github.com/blog/1184-contributing-guidelines

If you want to make it even more visible than that, you could add an ISSUE_TEMPLATE.md:
https://help.github.com/articles/creating-an-issue-template-for-your-repository/

ZinggJM, 2018-02-05:
@per1234 
Thank you for helping me with these links.
I will have to learn more about using GitHub; there is much information for me to discover.

closed 2018-02-05 by ZinggJM

per1234, 2018-06-13:
The URL is correct but the markup used to linkify it was bad. This one will work for you @mh00h :
https://forum.arduino.cc/index.php?topic=487007.330

---
### Some info on the cause of the link problem

The bad markdown looks like this:
```markdown
[https://forum.arduino.cc/index.php?topic=487007.330](url)
```
which causes the link to point to the nonexistent:
https://github.com/ZinggJM/GxEPD/issues/url

in fact GitHub automatically linkifys URLs so you can just do this:
```
https://forum.arduino.cc/index.php?topic=487007.330
```
If you do want to use Markdown's link markup it would look like this:
```markdown
[https://forum.arduino.cc/index.php?topic=487007.330](https://forum.arduino.cc/index.php?topic=487007.330)
```
That markup is more useful if you want to linkify text like this:
```markdown
The preferred location for questions and suggestions for enhancement is the ["Waveshare e-paper displays with SPI" thread](https://forum.arduino.cc/index.php?topic=487007.330) on the Arduino Forum.
```
which renders as:

>The preferred location for questions and suggestions for enhancement is the ["Waveshare e-paper displays with SPI" thread](https://forum.arduino.cc/index.php?topic=487007.330) on the Arduino Forum.