ZinggJM/GxEPD #31: Only first display update works: Firebeetle ESP32 / 4.2" Waveshare

eschlenz, 2018-03-07:
Thank you for this great library!

I have a Firebeetle ESP32 with the 4.2" Waveshare. I'm using the GxGDEW042T2 implementation. The display only updates the first time, and then refuses to update after that. Per #19, if I call `powerDown` before the 2nd update, then it works. I also saw the solution to #6, but I've checked my connections over 3 times and believe I have things connected properly.

It is also worth noting, that I _had_ been using the 1.54" Waveshare library with the GxGDEP015OC1 implementation and that worked without any issues. Same code.

I've created a code snippet that illustrates the issue. In the `loop` method, you will see a couple of lines that are commented out. If I uncomment those, then the screen updates as expected. Otherwise, it just doesn't:

```
#include <GxEPD.h>
#include <GxGDEW042T2/GxGDEW042T2.cpp>
#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>
#include <Fonts/FreeSansBold24pt7b.h>

GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 16, 4);
int i = 0;

// 'rain', 56x56px
const unsigned char rain [] PROGMEM = {
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x07, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0x80, 0x01, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 
  0xfe, 0x00, 0x00, 0x3f, 0xff, 0xff, 0xff, 0xfe, 0x00, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xfc, 0x00, 
  0x00, 0x00, 0x1f, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x0f, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 
  0x0f, 0xff, 0xff, 0xf8, 0x00, 0x00, 0x00, 0x07, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x07, 0xff, 
  0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0f, 0xf8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0xf0, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x03, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xc0, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x01, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x01, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xe0, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xc0, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x01, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0xf8, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x07, 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xef, 0xfb, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0x8f, 0xe3, 0xff, 0xff, 0xff, 0xff, 0xfc, 0x0f, 0xc3, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x0f, 
  0x83, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x0f, 0x83, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x0f, 0x83, 0xff, 
  0xff, 0xff, 0xff, 0xfe, 0x0f, 0x83, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xcf, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfb, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xe3, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x83, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x83, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0x83, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x83, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0x87, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf
};

void setup() {
  display.init();

  display.drawExampleBitmap(rain, 0, 0, 56, 56, GxEPD_BLACK);
  display.update();
  
  delay(2000);
}

void loop() {
  i++;
  
// After first update, everything works as long as I uncomment these two lines.
//  display.powerDown();
//  display.init();
  
  const GFXfont* f = &FreeSansBold24pt7b;

  uint16_t box_x = 60;
  uint16_t box_y = 60;
  uint16_t box_w = 90;
  uint16_t box_h = 100;
  uint16_t cursor_y = box_y + 16;
  
  display.setFont(f);
  display.setTextColor(GxEPD_BLACK);

  display.fillRect(box_x, box_y, box_w, box_h, GxEPD_WHITE);
  display.setCursor(box_x, cursor_y+38);
  display.print(i); 
  display.updateWindow(box_x, box_y, box_w, box_h, true);
  
  delay(5000);
}
```

ZinggJM, 2018-03-07:
This is a known issue with the 4.2" b/w display.
It is solved in the PartialUpdateExample by drawing the background twice: first using full update, then using partial update to full screen.
The solution would be to do this automatically after first power up, like I do in GxEPD2.
I keep this issue open, but it is solved in the example, and may not get solved general soon.

ZinggJM, 2018-03-28:
The described solution is implemented. More tests are pending, as this may not work for all cases, e.g. after several different full updates (to be verified).

z3tof, 2018-03-30:
Hello,

I have a Lolin 32 Lite ESP32 with the 4.2" Waveshare. I'm using the GxGDEW042T2 implementation.
It worked perfectly with older version.
Since the last version, it doesn't work as well : the first screen is refreshed twice.

Christophe

ZinggJM, 2018-03-30:
This is exactly the modification I applied for partial update to work if the application draws the background only once. I consider this the smaller disadvantage, as this happens only once after reset or program upload. As I kept this issue open, I may try to find e better solution as soon as I have time for this.
Of course you can initialize _initial to false in the constructor code, if this annoys you.

z3tof, 2018-03-31:
Thank for your answer ! and your library too !!! ;-)
This is a little ennoying for me because I do a deepsleep and each time it wakes up, it does an init()...
I will see your solution with _initial to false in the constructor code...

capofalcone, 2018-05-03:
Hi, I have the same problem, with an esp32 that goes to sleep and updates the display every 5 minutes. I tried by setting _initial to false, but nothing changed.
Is there a way to avoid the first double refresh?
Thank you so much
marco

ZinggJM, 2018-05-04:
Ok, there is a way to fix it, I just need to shift my priorities and do it. I am working on a new version of GxEPD2, and just need to copy the new solution from there, and test it. Sorry for the delay.

ZinggJM, 2018-05-04:
The double full refresh after reset is replaced by one full refresh followed by one full screen partial refresh. The partial refresh is not noticed, because nothing changes. The double refresh is needed for the case where the next refresh is a partial refresh with different content; old buffer content would show up otherwise.

capofalcone, 2018-05-07:
Thank you! It works like a charm now.
Great work 👍

closed 2018-07-26 by ZinggJM