ZinggJM/GxEPD #34: GxGDEW075Z09 broken

teo1978, 2018-03-24:
I tested the examples SPI_TestExample and PageDisplayExampleSmallRam on a WaveShare 7.5-inch b/w/r display
https://www.waveshare.com/7.5inch-e-paper-hat-b.htm
with its own "hat" and a Sparkfun ESP8266 Thing Dev controller.

When using the  GxGDEW075T8 class (b/w only), I get what looks like expected output, using the whole screen.

When using the GxGDEW075Z09 class, I get apparently sensible output only on the top half of the screen, while the bottom half is always filled with a single color (sometimes white, sometimes black, sometimes red).

I posted some photos here: https://forum.arduino.cc/index.php?topic=487007.msg3661797#msg3661797

ZinggJM, 2018-03-24:
If you want to sponsor me, you can provide me such a e-paper display.
Or you can contact the Contributor of the code for this display.
Or you can test GxEPD2_32 with this display, and tell me if it works. The buffer space issue is solved there by picture loop / paged drawing.

ZinggJM, 2018-03-24:
I do not know if its better to leave it as it is, or remove it from this repository.
But I will not fix it until I get such a e-paper for free. And if I fix it, it will not work on ESP8266 because of not enough RAM for full buffer. It might work on ESP32 and some STM32.

closed 2018-03-24 by ZinggJM

teo1978, 2018-03-24:
> And if I fix it, it will not work on ESP8266 because of not enough RAM for full buffer

I haven't yet investigated how the "PageDisplayExampleSmallRAM" example, but isn't that some sort of technique to allow using such big displays with devices with small RAM?

ZinggJM, 2018-03-24:
Yes, that's why I have implemented it in GxEPD2, but as I wrote, it is untested, as I don't have this display.

teo1978, 2018-03-24:
> Or you can test GxEPD2_32 with this display, and tell me if it works. 

It seems to work! (well at least I see some b/w/r output that occupies the whole screen)

Is there any documentation that explains what "simplified" means in "Simplified Version of GxEPD for 32 Bit Arduino"?

> I do not know if its better to leave it as it is, or remove it from this repository.

As long as you are not going to fix it, my humble suggestion is to remove it. You could leave a note on the readme or elsewhere, for contributors who may be interested in resuming the effort from where you left it.