ZinggJM/GxEPD #36: ESP3266 + screen 2.13 waveshare

Malewsm, 2018-03-28:
Hi,

I am currently using your libraries with the e-paper screen 2.13 by waveshare. The setup works really good, but there is one issue I cannot overcome. When I use the simplest code, for instance, to print on the screen a simple text every 5 seconds - each time before the screen refreshes, it goes black/white/back a few times (even when the text is the same as previously). Is it possible to avoid this? So that the screen will just refresh letters which are different?

Thank you for your help in advance,
Mateusz

ZinggJM, 2018-03-28:
> please use the Arduino Forum Displays for questions or enhancement suggestions

Please observe the guide in the issue_template.md.

This is **not** an issue of the library, but physical behavior of the e-paper display.
Black / white e-paper displays can support fast update for partial update, applying only the difference.
For this the controller internally has 2 buffers and uses a waveform table for partial update that handles only the differences. For 3-color e-paper displays this is not possible because of the physical process; the movement speed difference of the smaller black particles and the larger colored particles is used, which does not work using only differences, and the 2 buffers are used for black and colored.

closed 2018-03-28 by ZinggJM

G6EJD, 2018-03-28:
Epaper displays are really designed for updates occasionally not every 5-secs and they take a long time to refresh flashing many tines so what your seeing is completely normal. You need to rethink how frequently you update the display. Welcome to epaper displays and remember they take no power after refresh/update!

Malewsm, 2018-03-29:
Hey,   Thank you for your answer! Hmm, but in the project, I am working right now it is necessary to eliminate this flickering. It is possible with smaller screens! Is it a feature which somehow I can modify in software?   Thank you for your help,  Mateusz M

ZinggJM, 2018-03-29:
Search for "RGB e-paper", google finds. True RGB e-paper displays physically work like b/w e-paper, and thus can be fast as these. If you get me such a display, I will try to make a library for it.

G6EJD, 2018-03-29:
The smaller screen (1.54") is fast, about the same as the 2.9". If you want to get fast screen refresh times then an e-paper is not your solution. There are no software fixes you can apply as the screen delay is a physical limitation until the technology improves (if ever).

ZinggJM, 2018-03-29:
The main difference however is between b/w and 3-color e-paper displays. Waveshare has published a table with update times for all SPI e-papers they sell (on the page for the e-paper HAT).
And according to this table only the small b/w displays support partial update (without flashing).