ZinggJM/GxEPD #38: Waveshare 1.54inch_e-Paper_Module_(C) e-paper 3 colors

Pauperman66, 2018-04-22:
Used e-paper display model 1.54inch_e-Paper_Module_(C) B/W/Y with Bluepill-like board in combination with GxGDEW0154Z04 material. Even if all wires are ok and selected right headers display only flashes and font-based example and every other example produce only "funny" pictures. I thought to a different resolution between 2 models...

Does it work with Waveshare 3 colors equivalent 1.54" or is it different from Good-Disaplay one and it has to be developed a special purpose driver?

Pauperman66, 2018-04-24:
Right. Thanks for your quick answer. I will send result video (link) when I tried following code:

#include <GxEPD.h>

#include <GxGDEW0154Z04/GxGDEW0154Z04.cpp>  // 1.54" b/w/r

#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>
#include <Fonts/FreeMonoBold9pt7b.h>

// new mapping suggestion for STM32F1, e.g. STM32F103C8T6 "BluePill"
// it should work with Black Pill here too

// BUSY -> A1, RST -> A2, DC -> A3, CS-> A4, CLK -> A5, DIN -> A7

GxIO_Class io(SPI, /*CS=*/ SS, /*DC=*/ 3, /*RST=*/ 2);

GxEPD_Class display(io, /*RST=*/ 2, /*BUSY=*/ 1);

void setup(void)
{
  display.init();
}

void loop()
{
  display.drawPaged(page1);
  delay(4000);
  display.drawPaged(page2);
  delay(10000);
  display.drawPaged(page3);
  delay(10000);
}

void page2()
{
  const char* name = "FreeMonoBold9pt7b";
  const GFXfont* f = &FreeMonoBold9pt7b;
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(f);
  display.setCursor(0, 0);
  display.println();
  display.println("Hello there!");

}
void page3()
{
  const char* name = "FreeMonoBold9pt7b";
  const GFXfont* f = &FreeMonoBold9pt7b;
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(f);
  display.setCursor(0, 0);
  display.println();
  display.println("Doing well?!");

}

void page1()
{
  const char* name = "FreeMonoBold9pt7b";
  const GFXfont* f = &FreeMonoBold9pt7b;
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(f);
  display.setCursor(65, 65);
  display.println("RGDS, Alessandro");
}

Pauperman66, 2018-04-24:
video at link https://youtu.be/UZU24KtknHk

that is https://www.youtube.com/watch?v=UZU24KtknHk&feature=youtu.be

Pauperman66, 2018-04-24:
For instance Printing an upper case "A" only, instead of seen 1 character it is split in two dashed "A" apart each other

Pauperman66, 2018-04-24:
About resolution: into only program it works with this display (Waveshare program running under Arduino downloadable from link https://www.waveshare.com/w/upload/1/1c/1.54inch_e-paper_module_c_code.7z ) I see into header file named "epd1inch54.h" following statements:

// Display resolution
#define EPD_WIDTH       152
#define EPD_HEIGHT      152

Pauperman66, 2018-04-24:
I thought I did it writing here under "Issue". Nevertheless if I change into Library for item "GxGDEW0154Z04" into file "GxGDEW0154Z04.h" (original values were 200 for both)

#define GxGDEW0154Z04_WIDTH 160
#define GxGDEW0154Z04_HEIGHT 160

 can see something even if I do not know if character appearance is ok (it seems horizontally dashed)

Pauperman66, 2018-04-24:
Further update: with these changes above fonts are displayed properly but only using RED and WHITE (of course red means yellow on this display). Using BLACK no way

ZinggJM, 2018-04-25:
The 1.54" 152x152 3-color display uses IL0373, the 200x200 uses IL0376F. So just changing width and height does not work. The IL0373 is used in other displays as well, so I might even create a class for GxEPD for it without being able to test it. I will take a look at this in the next days or so.

Pauperman66, 2018-04-25:
Thank you so much for focusing your efforts on this. Looking forward to get updates from you. Take your time. In absence of posts in a "while" I will double check writing a friendly reminder.

ZinggJM, 2018-04-26:
I have added GxGDEW0154Z17 for 1.54" 3-color 152x152 display.
I need feedback to know if it really works on the 1.54" 3-color 152x152 panel, thank you.

ZinggJM, 2018-04-26:
Thank you for the feedback! With STM32F1 you should also have enough RAM to use GxEPD_SPI_TestExample.ino with this display.

closed 2018-04-26 by ZinggJM