ZinggJM/GxEPD #40: Heltec E-Paper 1,54" b/w does not work properly

knopserl, 2018-05-06:
I tried the first time this e-ink/e-paper:
https://www.aliexpress.com/item/1-54-Inch-E-Paper-Module-E-Ink-Display-Screen-Module-Black-White-Color-SPI-Support/32846836344.html?spm=a2g0s.9042311.0.0.LVN8Te

But I did not get it to work properly. It shows most of the time a "white noise" = random pixel and sometime suddenly it shows a proper text or some of the demo graphics. 

It has only the minimum SPI pins like SI, CLK, CS as well as D/C but no busy and reset.
If I use the waveshare 1,54" e-paper with busy and reset it works.

I have set the pins to -1 or ommited the pin in the class instances but it did not change the behavior.
I've tried other e-paper classes but it got worse so the following class seem to be the best as it shows at least someting or sometimes (rarely) correct
#include <GxGDEP015OC1/GxGDEP015OC1.cpp>    // 1.54" b/w
GxIO_Class io(SPI, /*CS=D8*/ SS, /*DC=D3*/ D3 , /*RST=D4*/ -1); // arbitrary selection of D3(=0), D4(=2), selected for default of GxEPD_Class
GxEPD_Class display(io, /*RST=D4*/ -1, /*BUSY=D2*/ -1); // default selection of D4(=2), D2(=4)

I've tried with STM32 and ESP8266 Weos D1 R2 Mini and Mini Pro boards.
Any hints? Is this e-paper without Budy and Reset even supported yet?
I liked it more because it is smaller/thinner in size and has less pins to connect and it's cheaper and looks nice.

G6EJD, 2018-05-06:
Try putting a delay after the committ / screen update line so the display has time to update before it gets more data to display, maybe delay(3000) this will simulate the busy function

knopserl, 2018-05-06:
the code (your sample code) consist already of almost a delay of 5000 after each display command

  display.drawExampleBitmap(BitmapExample1, sizeof(BitmapExample1));
  delay(2000);
  display.drawExampleBitmap(BitmapExample2, sizeof(BitmapExample2));
  delay(5000);
  display.fillScreen(GxEPD_WHITE);
  display.drawExampleBitmap(BitmapExample1, 0, 0, GxEPD_WIDTH, GxEPD_HEIGHT, GxEPD_BLACK);
  display.update();
  delay(5000);

knopserl, 2018-05-06:
I have now a delay of at least 3000ms after each display command and after each display.update and it does not  help for this e-paper. There is no help/lib/hint for that display only the info that it used a IL3829 controller chip.

BTW, I'M using your GxEPD_SPI_TestExample exaple on a Wemos D1 R2 Mini and D1 Mini Pro.

ZinggJM, 2018-05-08:
I had noticed these new e-paper boards, which seem to have panels from Good Display, but no BUSY pin.
It is the vendors obligation to provide suitable driver or demo code for the devices he sells.
I am willing to add support for known e-paper panels (from Good Display) to GxEPD and GxEPD2, if I do not need to buy the display boards myself (Sponsors Are Welcome), as I need to be able to test.

xmod13, 2018-05-09:
Hi,
Got the same issue here with this EPD. Partial update works from time to time only .
I contacted supplier and he gave me this link with his library+example code : 
http://www.mediafire.com/file/2egqxyegc74tfso/%E7%94%B5%E7%BA%B8%E5%B1%8F%E7%A8%8B%E5%BA%8F%E8%B5%84%E6%96%99.rar
I will give it a try.

That said, if you guys find a solution using GxEPD library, I would be more than interested!

knopserl, 2018-05-09:
@ZinggJM and @xmod13 
I also contacted the E-Ink vendor and got the same link yesterday. Today I've tried the demo and it works.
I had to do some small changes as I'm using a Wemos Mni D1 with an ESP8266 MCU and the code does not support this MCU. I also had to change the pin definitions

So here are the two changes:
epdif.h:
// Pin definition
#define RST_PIN         -1     // no reset pin
#define DC_PIN          D4
#define CS_PIN          D8
#define BUSY_PIN       -1  // no busy pin

in each cpp or h file where 
#include <avr\pgmspace.h>
is used, I changed it to a more cross MCU compliant code:

#if (defined(__AVR__))
#include <avr\pgmspace.h>
#else
#include <pgmspace.h>
#endif

I have attaxhed the resulting picture.

Maybe ZinggJM can derive the required changed for GxEPD out of the provided library files provided by the library: I have attched my modified files in this post 
[E-Ink_1.54.zip](https://github.com/ZinggJM/GxEPD/files/1989442/E-Ink_1.54.zip)
Here the pricture

[image]

ZinggJM, 2018-05-10:
No, I don't want to apply changes I can't tests with the device the changes are for.
Tell the vendor to send me these displays, if he is interested in support of GxEPD for his displays.

knopserl, 2018-05-10:
I have send an inquity to Heltec requesting a free E-Ink display for you. Let's see what happens.
I hope they will respond to my opened issue thread and will get in contact with you.
Otherwise, I'm not sure if I'll order more of this displays though they are quite nice (very thin, same e-paper as waveshare, less required pins, cheaper)

ZinggJM, 2018-05-16:
I have added "no BUSY" support to GxGDEP015OC1. @knopserl may test it with his display.
I may add this support to additional classes if I get sponsored with these displays.
Kept "open" until got confirmed that it works.

knopserl, 2018-05-16:
@ZinggJM I have tested it and it seem to work. Great job!
I will do some more tests like partial update and will give more feedback soon.

ZinggJM, 2018-05-17:
This is only implemented for the 1.54" b/w e-paper display. Thanks to @knopserl and his request to Heltec. Let's see if I get some of these displays.

knopserl, 2018-05-17:
@ZinggJM right it is only implemented for the 1.54" b/w paper not for all the others.
I hope Heltec is providng you some displays. I'll send them a reminder and statement again, that I will not buy any further displays from them if they don't support your and your library. I don't really like their implementation. It is not even ready for the great ESP MCU's (I had to do the adaptations by myself).
Anyhow thats for at least the 1,54" b/w aas these are the ones I own from Heltec. All others are from Waveshare.

pantale, 2018-07-20:
Hi

I also have ordered this e-ink screen but I'm unable to make it work with an esp32 devkit.

Olivier

knopserl, 2018-07-21:
@pantale I use the Heltec E-Paper on SPI 2 as I use SPI 1 for the Radio CC1101.
You can/should use it on the first/default SPI 1. I use all the pins of the Heltec as it requires all but hthere is nothing special I'm using:
[image]

Here is the code snipped of my sketch
#include <GxEPD.h>
#include <GxGDEP015OC1/GxGDEP015OC1.cpp>    // 1.54" b/w
#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>

// FreeFonts from Adafruit_GFX
#include <Fonts/FreeSans9pt7b.h>
#include <Fonts/FreeSans12pt7b.h>
#include <Fonts/FreeMono9pt7b.h>
#include <Fonts/FreeMonoOblique9pt7b.h>
#include <Fonts/FreeMonoBold9pt7b.h>
#include <Fonts/FreeMono12pt7b.h>
#include <Fonts/FreeMonoBold12pt7b.h>
#include <Fonts/FreeMonoBold18pt7b.h>
#include <Fonts/FreeMonoBold24pt7b.h>

#define _spi1_cs   17  // goes to radio CS
#define _spi1_data 16  // goes to radio Data
#define _spi1_mosi 23  // goes to radio MOSI
#define _spi1_miso 19  // goes to radio MOSI
#define _spi1_sclk 18  // goes to radio SCK/CLK
//
#define _spi2_cs   26  // goes to display CS
#define _spi2_dc   27  // goes to display DC
#define _spi2_mosi 13  // goes to display MOSI
#define _spi2_miso 12  // goes to display MOSI
#define _spi2_sclk 14  // goes to display SCK/CLK

#define epd_bsy    -1
#define epd_rst    -1

SPIClass SPI2(2);
GxIO_Class io(SPI2, _spi2_cs, _spi2_dc, epd_rst, epd_bsy);
GxEPD_Class display(io, epd_rst, epd_bsy);
.....
void initDisplay()
{
  SPI2.begin(_spi2_sclk, _spi2_miso, _spi2_mosi, _spi2_cs); //CLK,MISO,MOSI,SS
  display.init();
  display.setRotation(1);
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.update();  
}

pantale, 2018-07-23:
Yes @knopserl
It works now with your help.
Thanks a lot.

ZinggJM, 2018-07-26:
Support for the Heltec 1.54" e-paper is kept in GxEPD, although I didn't get sponsored with any of these BUSY-less e-papers. General support for Good Display SPI e-papers on boards without BUSY connection is available in GxEPD2.

closed 2018-07-26 by ZinggJM

knopserl, 2018-07-26:
@ZinggJM a big thanks to you!

mathis92, 2018-10-15:
Does anybody get this Heltec e-paper working with nodeMcu ESP8266? i am struggling with running examples. Not sure if the wiring is ok currently my wiring is 
D3 - > DC
D5 -> CLK 
D7 -> SDI
D8 -> CS
VCC -> VCC
GND -> GND 
Does anybode have any suggestion or example how to make it work ? 
Thanks a lot

G6EJD, 2018-10-15:
Have you tried the code example above from knopserl change the pins to your allocations which are ok and it should work, you may have memory problems (lack of) with the esp8266.

ZinggJM, 2018-10-16:
Do you use a NodeMcu 3.0 ?

Then you might take a look at this Arduino Forum post: https://forum.arduino.cc/index.php?topic=487007.msg3908242#msg3908242