ZinggJM/GxEPD #42: 7.5bwr (GxGDEW075Z09) connected to ESP32 causes boot loop

daduke, 2018-05-14:
Hi,

I followed https://www.heise.de/select/ct/2018/2/1515453575487133 to build an e-paper display connected to an ESP32. I use the Waveshare 7.5bwr. Just like https://www.heise.de/forum/c-t/Kommentare-zu-c-t-Artikeln/Drahtloses-Tuerschild-mit-E-Paper-Display/Fehler-bei-7-5bwr/posting-31883746/show/ I experience a boot loop when specifying 7.5bwr, but 7.5 works. In the forum they suggest Basecamp (an ESP32 convenience lib) be at fault, but I can rule this out since a) it doesn't for for either Basecamp version and b) the crash also happens without basecamp at all.

### E-Paper module used
https://www.waveshare.com/7.5inch-e-Paper-C.htm

### Arduino / Processor module used
ESP32

what further information should I provide to debug this? The crash message is:

```
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c
assertion "res == pdTRUE" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c", line 375, function: start_cpu0_default
abort() was called at PC 0x400d8c27 on core 0

Backtrace: 0x4008dbec:0x3ffe3b90 0x4008ddc7:0x3ffe3bb0 0x400d8c27:0x3ffe3bd0 0x40082972:0x3ffe3c00 0x40082b10:0x3ffe3c30 0x40078892:0x3ffe3c50 0x400788f9:0x3ffe3c80 0x40078a55:0x3ffe3cb0 0x40078b7e:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
```

I have no idea where the /Users/ficeto/ stuff comes from...

thanks,
-d

ZinggJM, 2018-05-15:
Thank you for reporting this issue.

I believe I have tested the examples also with ESP32. But the code of the class for this display is incomplete.
yield() is done for ESP8266 in the drawPaged methods, but not for ESP32; if the drawCallback method takes a little longer, watchdog timeout may occur.

`#if defined(ESP8266)
    yield();
#endif
`
should be changed to

`#if defined(ESP8266) || defined(ESP32)
    yield();
#endif
`
in several places

I will fix this soon.

ZinggJM, 2018-05-15:
Please try to create a minimal example that shows the issue, and add it as zip file to your post, so I can try it myself. Does the GxEPD_SPI_TestExample work?

daduke, 2018-05-15:
GxEPD_SPI_TestExample works fine. Here's the zip with the minimal example with Basecamp stripped out.

thanks!
[waveshare.zip](https://github.com/ZinggJM/GxEPD/files/2006166/waveshare.zip)

ZinggJM, 2018-05-15:
Thank you, I will try to analyze it tomorrow.

ZinggJM, 2018-05-16:
The minimal example is not minimal enough for me. You need at least to provide a link to the library AsyncTCP, as it is not available through Library Manager.

daduke, 2018-05-16:
sorry about that: https://github.com/me-no-dev/AsyncTCP

ZinggJM, 2018-05-16:
There seems to be an error in line 153 : bitRead for a negative bit number. Try with bool hasRed = false;

ZinggJM, 2018-05-16:
It works with my 7.5" b/w/r e-paper display, showing "Could not connect to".
I can't test with pictures, as it needs a server for this.
Drawing big colored pictures pixel by pixel may be the cause of the issue, this takes time and could cause watchdog timeout. I am not familiar with exception codes of ESP32. Adding some yield() in the drawPixels method may help.

ZinggJM, 2018-05-17:
I also use Arduino 1.8.5, but with Wemos LOLIN32 (LOLIN32 Lite device). I tested on 2 notebooks, one with earlier Espressif ESP32 package, one with the current one. Sending HEX file for different target doesn't make sense.
You could change the SSID in your code, to see if you also get "Could not connect to".
And you can add debug output (Serial.print) to find out where it causes the exception.
Use e.g. display.init(115200) to get diagnostic output from GxEPD.

daduke, 2018-05-17:
I'm afraid I cannot really follow your reasoning, but commenting out everything in onDataHandler() doesn't change a thing, so that's not it.

ZinggJM, 2018-05-17:
No problem. The problem is with the Espressif compiler. The 7.5" 3-color display needs too much RAM for your example, combined with the compiler issue.
What you can do is change line 35 from
`#define GxGDEW075Z09_PAGES 1 // e.g. ESP32
`
to
`#define GxGDEW075Z09_PAGES 2 // e.g. ESP32
`
in GxGDEW075Z09.h

daduke, 2018-05-17:
this works, but seems to cut the display in half...

ZinggJM, 2018-05-18:
Yes, if we half the buffer size, we can only draw half the display with update().
To use full display with half (or less) buffer size, we need to use paged drawing, but paged drawing is not well suited for use with bitmap download. GxEPD2_32 has methods to write bitmap parts or full bitmaps to controller memory, and then refresh the display.

> this works, but

So we have proof the issue is with RAM usage in the ESP32 package. One of the current 340 issues.
It may be related to this issue: [https://github.com/espressif/arduino-esp32/issues/1163](url)

closed 2018-05-18 by ZinggJM