ZinggJM/GxEPD #43: Compiling Error when using Example (Font related)

MortimerNeuss, 2018-05-21:
Hello

I tried to use the  GxEPD_SPI_TestExample with Arduino IDE 1.8.2 with Generic ESP8266 Module (Wemos D1 mini Pro) but the compiler failed. Im about to use the Waveshare 2.9" b/w E-Paper Display (so I include the GxGDEH029A1/GxGDEH029A1.cpp)
I copied the Font Folder from the Adafruit-GFX-Library-master into the branch of this repo. The description on how and where to put the Folder are not detailed, so this might trigger the error.

The Error log reads:

```
In file included from \Arduino\libraries\GxEPD/GxEPD.h:20:0,
                 from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:35:

\Arduino\libraries\GxEPD/GxFont_GFX.h:37:24: error: 'GFXfont' does not name a type
     void setFont(const GFXfont *f = NULL);
                        ^

In file included from \Arduino\libraries\GxEPD/GxEPD.h:19:0,
                 from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:35:

\Arduino\libraries\Adafruit_GFX/Adafruit_GFX.h:11:20: error: expected unqualified-id before '{' token
 #define swap(a, b) { int16_t t = a; a = b; b = t; }
                    ^

\Arduino\libraries\GxEPD/GxGDEH029A1/GxGDEH029A1.h:74:5: note: in expansion of macro 'swap'
     swap(T& a, T& b)
     ^

In file included from \Arduino\libraries\GxEPD/GxGDEH029A1/GxGDEH029A1.cpp:14:0,
                 from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:43:

\Arduino\libraries\GxEPD/GxGDEH029A1/GxGDEH029A1.h:75:5: error: expected unqualified-id before '{' token
     {
     ^

In file included from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:56:0:

\Arduino\libraries\GxEPD/Fonts/FreeMonoBold9pt7b.h:87:7: error: 'GFXglyph' does not name a type
 const GFXglyph FreeMonoBold9pt7bGlyphs[] PROGMEM = {
       ^

\Arduino\libraries\GxEPD/Fonts/FreeMonoBold9pt7b.h:184:7: error: 'GFXfont' does not name a type
 const GFXfont FreeMonoBold9pt7b PROGMEM = {
       ^

In file included from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:57:0:
\Arduino\libraries\GxEPD/Fonts/FreeMonoBold12pt7b.h:148:7: error: 'GFXglyph' does not name a type
 const GFXglyph FreeMonoBold12pt7bGlyphs[] PROGMEM = {
       ^

\Arduino\libraries\GxEPD/Fonts/FreeMonoBold12pt7b.h:245:7: error: 'GFXfont' does not name a type
 const GFXfont FreeMonoBold12pt7b PROGMEM = {
       ^

In file included from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:58:0:
\Arduino\libraries\GxEPD/Fonts/FreeMonoBold18pt7b.h:321:7: error: 'GFXglyph' does not name a type
 const GFXglyph FreeMonoBold18pt7bGlyphs[] PROGMEM = {
       ^

\Arduino\libraries\GxEPD/Fonts/FreeMonoBold18pt7b.h:418:7: error: 'GFXfont' does not name a type
 const GFXfont FreeMonoBold18pt7b PROGMEM = {
       ^

In file included from \Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino:59:0:
\Arduino\libraries\GxEPD/Fonts/FreeMonoBold24pt7b.h:570:7: error: 'GFXglyph' does not name a type
 const GFXglyph FreeMonoBold24pt7bGlyphs[] PROGMEM = {
       ^

\Arduino\libraries\GxEPD/Fonts/FreeMonoBold24pt7b.h:667:7: error: 'GFXfont' does not name a type
 const GFXfont FreeMonoBold24pt7b PROGMEM = {
       ^

GxEPD_SPI_TestExample:445: error: 'GFXfont' does not name a type
 void showFont(const char name[], const GFXfont* f)

                                        ^

GxEPD_SPI_TestExample:445: error: ISO C++ forbids declaration of 'f' with no type [-fpermissive]
 void showFont(const char name[], const GFXfont* f)

                                                 ^

\Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino: In function 'void loop()':
GxEPD_SPI_TestExample:178: error: 'FreeMonoBold9pt7b' was not declared in this scope
   showFont("FreeMonoBold9pt7b", &FreeMonoBold9pt7b);
                                  ^

GxEPD_SPI_TestExample:179: error: 'FreeMonoBold12pt7b' was not declared in this scope
   showFont("FreeMonoBold12pt7b", &FreeMonoBold12pt7b);
                                   ^

\Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino: At global scope:
GxEPD_SPI_TestExample:445: error: 'GFXfont' does not name a type
 void showFont(const char name[], const GFXfont* f)
                                        ^

GxEPD_SPI_TestExample:445: error: ISO C++ forbids declaration of 'f' with no type [-fpermissive]
 void showFont(const char name[], const GFXfont* f)
                                                 ^

\Temp\arduino_modified_sketch_618414\GxEPD_SPI_TestExample.ino: In function 'void showFontCallback()':
GxEPD_SPI_TestExample:469: error: 'GFXfont' does not name a type
   const GFXfont* f = &FreeMonoBold9pt7b;
         ^

GxEPD_SPI_TestExample:472: error: 'f' was not declared in this scope
   display.setFont(f);
                   ^

exit status 1
'GFXfont' does not name a type
```

I tried to comment out everything related to the font topic (showFont & showFontCallback Functions) but then the compiler struggles with the GxEPD.h as it is still utilizing GxFont_GFX.h.

The Fonts are placed here:
\Arduino\libraries\GxEPD\Fonts\FreeMono(...).h

What have I missed? 

cheers
Mortimer

G6EJD, 2018-05-21:
Why not install adafruit gfx and leave as intended and fonts are expected in the adafruit fonts folder not the EPD folder.

ZinggJM, 2018-05-22:
No problem, I will delete this issue.

closed 2018-05-22 by ZinggJM