ZinggJM/GxEPD #44: GxEPD_SD_BitmapExample -> File not found

j0k3r187, 2018-06-28:
Hello from Germany

First of all thank you for this Library ;).

But when i try to test your Bitmap Example, i always get "File not found" on Serial Monitor.
Display works fine in your "GxEPD_SPI_TestExample" so my wiring is correct.

When i try a Test Scetch from SD Library that only looks for the file on Sd Card, Serial Monitor output

 Initializing SD card...initialization done.
 betty_1.BMP exists.

Im using a Pro Micro as Controller, the Waveshare 4.2 and SD Card Module.

Hope u can find the Issue in ur GxEPD_SD_BitmapExample

Regards

G6EJD, 2018-06-28:
Have you independently tested the SD-Card reader of your board as I have found it exceptionally difficult to get working with that board. It's down to the SPI pins and CS. Then when you have that running, you need to merge the connections for the display SPI bus and of course the chip select pins will be different. Hope this helps.

G6EJD, 2018-06-28:
What I would do is combine the two programmes, then comment out the SD elements and test the display, then add back in the SD library elements gradually to see where the programme starts to not read the card.  e.g. SD.begin(SD_CS) probably?

j0k3r187, 2018-06-28:
I tested this Method arleady in my Scetch and the Result was the same -> File not found. I added multiple 

if (SD.exists("betty_1.bmp")) {
    Serial.println(F("betty_1.BMP exists."));
  } else {
    Serial.println(F("betty_1.BMP doesn't exist."));
  }

and the issue starts after Initializing SD card... of the code.

But when i try an Example from an Library i thought it would work out of the Box.
The "GxEPD_SPI_TestExample" works fine so i think there is something wrong in the Code of the
"GxEPD_SD_BitmapExample"

ZinggJM, 2018-06-29:
I will look at this issue. I think I tested only with ESP8266 (Wemos D1 mini), as I had no SD module, and I stopped trying with the SD part of a TFT as soon I got the RobotDyn SD shield for D1 mini. Now I have separate DS modules; need to wire one to my 3.3V Arduino Pro Mini.
Maybe SD use with hardware SPI together with GxEPD will require SPI transaction support in GxEPD.

ZinggJM, 2018-07-03:
This issue is caused by insufficient RAM available for SD support. It does not compile for 2k AVR. It compiles for 2.5k AVR but has not enough free RAM for SD use (stack or heap size).
Library GxEPD2 now has an example GxEPD2_SD_AVR_Example.ino that uses the base display class without graphics buffer to work on AVR Arduino.

j0k3r187, 2018-07-04:
Hey ZinggJM,

nice Work and thanks for your Support. 
But if im reading correct the GxEPD2_SD_AVR_Example cant handle the Waveshare 4.2 b/w?

ZinggJM, 2018-07-04:
Yes, the Waveshare 4.2 b/w uses some more RAM for the Wavetable for fast partial update, just too much for a 2k AVR Arduino for this example. It might work on Pro Micro, I can't test, my Pro Micro is broken.

j0k3r187, 2018-07-04:
Hey ZinggJM,
you are my new Hero :). I tested ur GxEPD2_SD_AVR_Example with my Pro Micro and it works fine.
The Waveshare 4.2 b/w can display all bitmaps from ur GxEPD2 Library, of course the colered bitmaps will be recolored in b/w but it looks great.

Very much thanks from Germany for your Support

martinberlin, 2018-07-19:
@ZinggJM Quick question regarding this image topic:
It would be possible to have a image to display example with another image type that is not BMP ?
I'm trying to make an API example that will receive an URL and return the screenshot of the image to print in the display. The problem is that BMP for the display I'm using, a waveshare 7.5 " is too large, weights about 700 Kb. 
It would be possible to make an example for a PNG ?  I will try to read the image from the API, discard the headers and reproduce GxEPD_SD_BitmapExample but reading it from the web instead of the SD card. Thanks for the lib, works great for [me as well](https://www.youtube.com/watch?v=QxaL8tc7gh8)

ZinggJM, 2018-07-20:
GxEPD_SD_BitmapExample will be enhanced, e.g. additional bit depths, but not with other formats than BMP. For PNG search Arduino Forum, I saw some posts. For WiFi or SPIFFS support see GxEPD2 or GxEPD2_32, these are better suited for progressive write to the display.

martinberlin, 2018-07-24:
Alright, I have a working BMP 4 / 24 depth that reads the image from a stream. If it's Ok for you I will fork this and upload the example or copy somewhere the function. I could not make it work with 1 bit BMP don't know why.

ZinggJM, 2018-07-26:
Added GxEPD_SD_Example and GxEPD_WiFi_Example. GxEPD_SD_BitmapExample is removed, replaced with GxEPD_SD_Example, with more BMP depths supported. Not usable with small RAM AVR.
Note that GxEPD_WiFi_Example does not work reliable with ESP8266 with big bitmaps, because of a known issue with WiFiClientSecure of this package.

closed 2018-07-26 by ZinggJM