ZinggJM/GxEPD #45: 640x384, 7.5inch E-Ink display three-color flickers strange

webventil, 2018-07-03:
### Expected Behavior
Calling `display.update();` should update the display within some seconds.

### Actual Behavior
The display updates itself but than flickers for about ~10 seconds before finally updating the content.

### E-Paper module used
640x384, 7.5inch E-Ink display HAT for Raspberry Pi, yellow/black/white three-color

### Arduino / Processor module used
Espressif Systems ESP32-DevKit

### Steps to reproduce the behavior or reduced but complete code example
Run any example for `GxGDEW075Z09`. Calling `display.update()` will cause the flickering.
[image]

G6EJD, 2018-07-03:
Have you not used e-paper displays before? This is entirely normal operation especially for the colour variants.

webventil, 2018-07-03:
Thanks for your response. No these are my first steps with e-paper displays. I didn't know that. I'll close the issue.

closed 2018-07-03 by webventil

G6EJD, 2018-07-03:
It has to do this to remove polarisation of the particles, the it polarises yellow then black they are different sizes then the particle move to the display edge of you like and so an image is formed, it does take a while usually 4-5 secs, some of the smaller screens are very fast because there is less to refresh in terms of screen data and maybe just black elements.

ZinggJM, 2018-07-03:
If your video is in real-time, then your e-paper display updates much faster than my b/w/r 7.5" e-paper display. So it has a different waveform table in its controller OTP flash. This has also been reported in the Arduino Forum Displays.

webventil, 2018-07-03:
@G6EJD Thanks for taking the time to explain that. 
@ZinggJM I had to erase a bit from the end to be below the max file size limit from github. It takes in total ~19 seconds. That is indeed faster as the number waveform announces.

ZinggJM, 2018-07-03:
Ok, thanks for the clarification. Then it seems the same. You can look at the times reported in diagnostic output in Serial Monitor, the times in microseconds the BUSY line is active.