ZinggJM/GxEPD #49: Linker stops working when using U8g2 with GxEPD

42656e, 2018-08-18:
Hi,
in the changelogs I can see the information that GxEPD supports the U8g2_for_Adafruit_GFX fonts. So I tried that because I needed a font with the German Umlauts. But I ran into the following problems: everytime I use these imports the linker just dies:
`#include <U8g2_for_Adafruit_GFX.h>
#include <GxEPD.h>
#include <GxIO/GxIO.cpp>
#include <GxGDEP015OC1/GxGDEP015OC1.cpp>
#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>`

The error is very confusing:
`C:\Users\benja\AppData\Local\Temp\arduino_build_277199\sketch\sketch_aug15a.ino.cpp.o: In function `setup':

C:\Users\benja\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO.cpp:37: undefined reference to `GxFont_GFX::setFont(unsigned char const*)'

C:\Users\benja\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO.cpp:37: undefined reference to `GxFont_GFX::setCursor(short, short)'

C:\Users\benja\AppData\Local\Temp\arduino_build_277199\sketch\sketch_aug15a.ino.cpp.o: In function `loop':

C:\Users\benja\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO.cpp:37: undefined reference to `GxFont_GFX::setFont(unsigned char const*)'

C:\Users\benja\Documents\Arduino\libraries\GxEPD-master/GxIO/GxIO.cpp:37: undefined reference to `GxFont_GFX::setCursor(short, short)'

C:\Users\benja\AppData\Local\Temp\arduino_build_277199\sketch\sketch_aug15a.ino.cpp.o:(.rodata._ZTV12GxGDEP015OC1[_ZTV12GxGDEP015OC1]+0x8): undefined reference to `GxFont_GFX::write(unsigned char)'

collect2.exe: error: ld returned 1 exit status`

Am I even right with my try to use the U8g2 fonts? I've also tried to include the GxEPD libraries first but then the font can't be found.

ZinggJM, 2018-08-18:
> // for use of additional fonts you need to enable these in file GxFont_GFX.h

I provide an example to show use of additional fonts: GxFont_GFX_Example

in there you find the above quote.

Unfortunately adding additional fonts to GxEPD is not possible just in the application, as it is added as a middle layer in the inheritance tree.

Good Luck!

42656e, 2018-08-18:
Oh man, shame on me. I just have not seen this info :/ Its working now!
Thank you for your super fast help.

closed 2018-08-18 by 42656e