ZinggJM/GxEPD #50: How to compile the examples?

greboide, 2018-08-18:
Hello, im having a lot of dificulties to compile the example SPI, i am using platformio on linux, but its being very hard, i am using mbed framework and am including a custom Arduino.h, are there any general instructions to compile the library?
 
version=2.3.15

### E-Paper module used
epd 1,54bw waveshare
### Arduino / Processor module used
stm32f103c8, aka bluepill
### Steps to reproduce the behavior or reduced but complete code example

G6EJD, 2018-08-18:
In your compiler environment it looks like data types such as uint16_t and uint8_t have not been defined and need to be before the gfx include statements.

ZinggJM, 2018-08-20:
I would need to know more about your compile environment, is it a standard Arduino IDE on a Linux system, or something different?

greboide, 2018-08-20:
im using  https://platformio.org/, i am facing dificulties with the dependancies, is there some document about dependancies? like which libraries are you using? which specific versions?

greboide, 2018-08-20:
also why are you mixing mbed with arduino?

ZinggJM, 2018-08-21:
> A simple E-Paper display library with common base class and separate IO class for Arduino.

GxEPD is made for the Arduino IDE development environment. It doesn't claim to work with any other build environment. platformio looks interesting, I don't know it yet, but I have no time for this now.

Please consider to close this issue, as it doesn't seem to be an issue with the purpose of this library.

greboide, 2018-08-21:
sure, i got it to compile with platformio, the secret was using this library from here https://github.com/adafruit/Adafruit-GFX-Library instead of platformio library, sorry for any inconvenience and thank you for this great library!

greboide, 2018-08-21:
i was confused cause Adafruit_GFX uses mbed and this code is all geared towards Arduino, so i was facing some dependency hell

closed 2018-08-21 by greboide