ZinggJM/GxEPD #51: I2C port not working when using 7.5" e-paper display

blenheim, 2018-08-20:
Hello,
first of all thanks for your great library that makes easier to use EPD.
Let me explain the problem that I'm experiencing using Waveshare 7.5" b/w/y EPD with library "GxGDEW075Z09/GxGDEW075Z09.cpp" (rev. 2.3.15).
I've connected the EPD to a Wemos and the display works great with the demo sketch using the wiring below:
// BUSY -> D2, RST -> D4, DC -> D3, CS -> D8, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V

Then in order to use I2C port with BME280 Adafruit library that requires pin D1 and D2 to work, I had to switch the BUSY pin from D2 to D0 by changing the code as follow:

GxEPD_Class display(io /*RST=D4*/ /*BUSY=D0*/); // default selection of D4(=2), D2(=4)

Once I compile and upload the demo sketch the EPD no longer works.

PS: I've tried the same by using a Waveshare 4.2" b/w EPD and the readdressing of the BUSY line from pin D2 to D0 works great.

Thanks in advance for your help.

Best regards,
Emanuel

ZinggJM, 2018-08-20:
Did you take a look at diagnostic output in Serial Monitor?
Does only the display not work, or does the processor not start up? Some pins influence boot mode and should not be used as inputs; I think this is the case for D0.
I use D6 (MISO) for BUSY in my IoT example with sensor on I2C.

blenheim, 2018-08-20:
Hello,
the display works perfectly when using D2 = BUSY.
If I configure D6 for BUSY it doesn’t work; on the serial monitor the following message is shown:

 1384, room 16 
tail 8
chksum =setup done
Busy Timeout!

If I move the wire from D6 to D2 the display starts to work immediately. The strange thing is that the sketch is properly uploaded with D6 = BUSY:

GxEPD_Class display(io /*RST=D4*/ /*BUSY=D6*/); // default selection of D4(=2), D2(=4)

About the boot, I’m actually uploading the sketch with the display disconnected from the Wemos otherwise it locks after compiling the sketch.

PS: I canot find your IoT example among the sketches that comes with the library; where I can get it?

Thanks,
Emanuele

G6EJD, 2018-08-20:
D2 is also SDA on the I2C bus so you can’t use simultaneous SPI and I2C

ZinggJM, 2018-08-20:
GxIO_Class io(SPI, SS, D3, D4);
// D2 reserved for SDA, use D6 (MISO) instead
GxEPD_Class display(io, D4, D6);

in IoT_SHT31LP_Example_1.54inchEPD

blenheim, 2018-08-20:
@ZinggJM 
Yes, I've found the IoT sketch and I've tried D6 = BUSY but the display is still not working. 
I remark that the sketch properly compiles with D6 = BUSY but the display do not work with the BUSY pin of the display connected to D6. The very strange thing is that if I disconnect the wire from D6 and connect to D2 the display starts to work immediately, even if the sketch has been compiled with D6 = BUSY..

I add that with Waveshare 4.2" EPD there is no problem and I can configure both D0 or D6 as BUSY, allowing the full I2C functionality over D1 and D2.

ZinggJM, 2018-08-20:
Please post these lines as you have them written, to avoid this confusion!

blenheim, 2018-08-20:
The sketch is one among the demos that comes with the library; I have just selected the display type and I've configured the BUSY signal as D6. With this configuration the 7.5" display is not working.
Below I've attached the sketch as I'm using it now.

// GxEPD_SPI_TestExample : test example for e-Paper displays from Waveshare and from Dalian Good Display Inc.
//
// Created by Jean-Marc Zingg based on demo code from Good Display,
// available on http://www.good-display.com/download_list/downloadcategoryid=34&isMode=false.html
//
// The e-paper displays are available from:
//
// https://www.aliexpress.com/store/product/Wholesale-1-54inch-E-Ink-display-module-with-embedded-controller-200x200-Communicate-via-SPI-interface-Supports/216233_32824535312.html
//
// http://www.buy-lcd.com/index.php?route=product/product&path=2897_8363&product_id=35120
// or https://www.aliexpress.com/store/product/E001-1-54-inch-partial-refresh-Small-size-dot-matrix-e-paper-display/600281_32815089163.html
//

// Supporting Arduino Forum Topics:
// Waveshare e-paper displays with SPI: http://forum.arduino.cc/index.php?topic=487007.0
// Good Dispay ePaper for Arduino : https://forum.arduino.cc/index.php?topic=436411.0

// mapping suggestion from Waveshare SPI e-Paper to Wemos D1 mini
// BUSY -> D2, RST -> D4, DC -> D3, CS -> D8, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V

// mapping suggestion from Waveshare SPI e-Paper to generic ESP8266
// BUSY -> GPIO4, RST -> GPIO2, DC -> GPIO0, CS -> GPIO15, CLK -> GPIO14, DIN -> GPIO13, GND -> GND, 3.3V -> 3.3V

// mapping suggestion for ESP32, e.g. LOLIN32, see .../variants/.../pins_arduino.h for your board
// NOTE: there are variants with different pins for SPI ! CHECK SPI PINS OF YOUR BOARD
// BUSY -> 4, RST -> 16, DC -> 17, CS -> SS(5), CLK -> SCK(18), DIN -> MOSI(23), GND -> GND, 3.3V -> 3.3V

// new mapping suggestion for STM32F1, e.g. STM32F103C8T6 "BluePill"
// BUSY -> A1, RST -> A2, DC -> A3, CS-> A4, CLK -> A5, DIN -> A7

// mapping suggestion for AVR, UNO, NANO etc.
// BUSY -> 7, RST -> 9, DC -> 8, CS-> 10, CLK -> 13, DIN -> 11

// include library, include base class, make path known
#include <GxEPD.h>

// select the display class to use, only one
//#include <GxGDEP015OC1/GxGDEP015OC1.cpp>    // 1.54" b/w
//#include <GxGDEW0154Z04/GxGDEW0154Z04.cpp>  // 1.54" b/w/r 200x200
//#include <GxGDEW0154Z17/GxGDEW0154Z17.cpp>  // 1.54" b/w/r 152x152
//#include <GxGDE0213B1/GxGDE0213B1.cpp>      // 2.13" b/w
//#include <GxGDEW0213Z16/GxGDEW0213Z16.cpp>  // 2.13" b/w/r
//#include <GxGDEH029A1/GxGDEH029A1.cpp>      // 2.9" b/w
//#include <GxGDEW029Z10/GxGDEW029Z10.cpp>    // 2.9" b/w/r
//#include <GxGDEW027C44/GxGDEW027C44.cpp>    // 2.7" b/w/r
//#include <GxGDEW027W3/GxGDEW027W3.cpp>      // 2.7" b/w
//#include <GxGDEW042T2/GxGDEW042T2.cpp>      // 4.2" b/w
//#include <GxGDEW042Z15/GxGDEW042Z15.cpp>    // 4.2" b/w/r
//#include <GxGDEW0583T7/GxGDEW0583T7.cpp>    // 5.83" b/w
//#include <GxGDEW075T8/GxGDEW075T8.cpp>      // 7.5" b/w
#include <GxGDEW075Z09/GxGDEW075Z09.cpp>    // 7.5" b/w/r

#include GxEPD_BitmapExamples

// FreeFonts from Adafruit_GFX
#include <Fonts/FreeMonoBold9pt7b.h>
#include <Fonts/FreeMonoBold12pt7b.h>
#include <Fonts/FreeMonoBold18pt7b.h>
#include <Fonts/FreeMonoBold24pt7b.h>

#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>

#if defined(ESP8266)

// generic/common.h
//static const uint8_t SS    = 15; // D8
[... 443 lines cut ...]
  display.fillScreen(GxEPD_WHITE);
  display.drawExampleBitmap(gImage_IMG_0001, x, y, 64, 180, GxEPD_BLACK);
  display.update();
  delay(500);
  uint16_t forward = GxEPD::bm_invert | GxEPD::bm_flip_x;
  uint16_t reverse = GxEPD::bm_invert | GxEPD::bm_flip_x | GxEPD::bm_flip_y;
  for (; y + 180 + 5 <= display.height(); y += 5)
  {
    display.fillScreen(GxEPD_WHITE);
    display.drawExampleBitmap(gImage_IMG_0001, x, y, 64, 180, GxEPD_BLACK, forward);
    display.updateWindow(0, 0, display.width(), display.height());
    delay(500);
  }
  delay(1000);
  for (; y >= 5; y -= 5)
  {
    display.fillScreen(GxEPD_WHITE);
    display.drawExampleBitmap(gImage_IMG_0001, x, y, 64, 180, GxEPD_BLACK, reverse);
    display.updateWindow(0, 0, display.width(), display.height());
    delay(1000);
  }
  display.update();
  delay(1000);
}
#endif

blenheim, 2018-08-20:
PS: With the sketch above the only way to make it work is to leave the BUSY wire connected to D2.

ZinggJM, 2018-08-20:
posting without code marks modifies slash-star and star-slash unfortunately
this is the original:
```GxEPD_Class display(io /*RST=D4*/ /*BUSY=D2*/); // default selection of D4(=2), D2(=4)```
this is what you need:
```GxEPD_Class display(io, /*RST=D4*/ D4, /*BUSY=D2*/ D6); // selection of D4, D6```
modifying comment only has no effect!

closed 2018-08-21 by ZinggJM