ZinggJM/GxEPD #52: 2.9" B/W display - partial update problem

blenheim, 2018-08-28:
Hello, I'm experiencing problem using partial update on 2.9" Waveshare EPD since the display properly updates just the half left side of the display while the right side remains white. 
The display is wired to a LOLIN ESP8266 through SPI port  in order to display enviromental datas coming from a BME280 sensor connected through I2C port.

It happens when I'm using the following partial update command:

**display.updateWindow(0, 0, 128, 296, true);**
or
**display.updateWindow(0, 0, GxEPD_WIDTH, GxEPD_HEIGHT, true);**
(see picture below)
[image]

If I shift the X coordinate by 45 dots in the update window command:
**display.updateWindow(45, 0, 128, 296, true);**
the partial update windows shifts to the right, partially discovering the text that was hidden before, but loosing the text on the left. The partial update windows seems to be approximately 128x128 pixels in size; outside these limits the display does not updates (see picture below)
[image]

When replacing the partial update command with the full update command:
**display.update();**
the display properly shows the full contents of the display as expected, but unfortunately with the full refresh cycle.
(see picture below)
[image]

Don't really know if is an issue with the library of other; any help will be really appreciated.

Thanks in advance.

All the best,
Emanuel :)

G6EJD, 2018-08-28:
Have you tried reversing the 128 and 296 values, your column width looks like it’s 128 pixels wide to me. It could be screen rotation is not rotating the coordinates.

blenheim, 2018-08-29:
Yes, in this case the partial update doesn't work at all..

ZinggJM, 2018-08-29:
**You should observe the issue template when you issue an issue**
And you should be aware that there are 32 watchers of this repository that all get informed by mail about your post.
```display.updateWindow(0, 0, GxEPD_WIDTH, GxEPD_HEIGHT, true);```
is complete nonsense! rotating the full physical screen dimensions makes no sense, and does **not** update the full screen if rotation happens to be different from 0 or 2.
Try with 
```display.updateWindow(0, 0, GxEPD_WIDTH, GxEPD_HEIGHT, false);```
Re-open, if this doesn't fix your problem, but provide a reduced example that shows your error.

closed 2018-08-29 by ZinggJM

blenheim, 2018-08-29:
Hello, I apologise for not having followed the template; I will pay more attention for the future.
About my issue, since I did not exactly understand how the partial update command works, (eg. when to use "true" or "false")  I did some tests in order to figure it out. 
**Now  I've revised the sketch according to your suggestion and it works fine. Many thanks.**
One question: is there some documentation other than the sample sketches that briefly explains all the library commands and their related funcionality?

**Thanks for your great job.**
Emanuel

ZinggJM, 2018-08-30:
>    ```// partial update of rectangle from buffer to screen, does not power off```
>   ```void updateWindow(uint16_t x, uint16_t y, uint16_t w, uint16_t h, bool using_rotation = true);```

Documentation is not my strength. It looks like most users of GxEPD can live with that.

Header files are best place to look for information, as these stay up-to-date in general.
I try to use meaningful names for methods and parameters.