ZinggJM/GxEPD #54: Documentation ?

CorentinPacaud, 2018-09-09:
Hello,

This is not an issue but a request.

Can you write a documentation of your lib ?

I mean, in the partialUpdateTest, there is those two function :

  display.fillRect(box_x, box_y, box_w, box_h, GxEPD_WHITE);
  display.updateWindow(box_x, box_y, box_w, box_h, true);

What are the meaning of those params ? GxEPD_WHITE for the first one and true for the second one ?

Same for this one : 
  display.setRotation(1);

What should I put for 0, 90, 180, ... ?

If the doc is already written, please provide the link in the ReadMe.

Thanks.

CorentinPacaud, 2018-09-09:
OK, 
How do I know that setRotation(1) will turn clockwise or anticlockwise ? Or maybe "1" is for 1° ?
I mean, a documentation like Android does or any others libs (not in arduino) do it.
It easy really hard and time consumming just figuring out how "this" function works. 2 lines of doc would save hours of work in total for everyone.

ZinggJM, 2018-09-09:
There is some comments in the header files, and Adafruit_GFX has its own documentation.

**if you don't like it, don't use it. simple enough**

closed 2018-09-09 by ZinggJM

G6EJD, 2018-09-09:
The rotation is easy, like the gfx variant, each increment rotates through 90° clockwise