ZinggJM/GxEPD #55: Partial Refresh?

novusabeo, 2018-09-10:
## Please post only issues that are real issues with this library!
- please use the Arduino Forum Displays for questions or enhancement suggestions

### Version number of the library used, should be the current one
- this number is needed for later reference and tracking

### Partial Refresh

### Full Refresh after anything 

### Using GxGDEW029Z10

### Powered from TTGO ESP32

### Steps to reproduce the behavior or reduced but complete code example

ZinggJM, 2018-09-10:
**version number is required for reference (history)!** Actual version is 2.3.16
Partial update works with my GDEW029Z10, only the window part is updated and "flashes".
3-color displays physically can't support "fast partial update" aka differential update.

closed 2018-09-10 by ZinggJM

novusabeo, 2018-09-10:
Oh I see. Thank you.

novusabeo, 2018-09-10:
What if I was to code it as if it was just a b/w display?? Say the GxGDEH029A1?

ZinggJM, 2018-09-10:
**please observe the issue template!** the issue is closed, and issues in GxEPD **are not to be used** for discussions.