ZinggJM/GxEPD #56: wdt reset - Waveshare 7.5" + Driver Board (ESP8266)

SunboX, 2018-10-01:
Hi, I'm using a [Waveshare 7.5inch e-Paper Display (Black and White)](https://www.waveshare.com/wiki/File:7.5inch-e-paper-specification.pdf) together with a [Waveshare e-Paper ESP866 Driver Board (ESP8266)](https://www.waveshare.com/wiki/E-Paper_ESP8266_Driver_Board).

I build a small Sketch using `ESP8266WiFi`, `WiFiClientSecure`, the `ArduinoJson` lib and `GxGDEW075T8`.

I'm running on OSX. These are my build settings:

[image]

Versions I use:

* Board: **esp8266** by **ESP8266 Community** Version **2.4.2**
* **Adafruit GFX Library** by **Adafruit** Version **1.2.9**
* **GxEPD** by **Jean-Marc Zingg** Version **2.3.15**
* **ArduinoJson** by **Benoit Blanchon** Version **5.13.2**

After doing a WiFi connection and fetching the [JSON](https://api.opensensemap.org/boxes/5b28d3891fef04001bd7dfae), I try to show some information on the display.
All is running in the `loop()` but is wrapped in a `function()` call. So, there are not much globals.

Now I'm getting a hardware (!) wdt reset directly inside `display.init()` of `GxGDEW075T8.cpp`. I nailed it down to the [`GxGDEW075T8::fillScreen`](https://github.com/ZinggJM/GxEPD/blob/master/src/GxGDEW075T8/GxGDEW075T8.cpp#L97) function.

I tried already several things to fix this issue. But nothing really helped. So currently I'm out of ideas and appreciate any help.
First, I tried to `yield()` and put a `ESP.getFreeHeap()` into the `fillScreen(...)` method. This is the change and the output:

```c++
void GxGDEW075T8::fillScreen(uint16_t color)
{
  Serial.println("fillScreen enter");

  uint8_t data = (color == GxEPD_BLACK) ? 0xFF : 0x00;
  for (uint16_t x = 0; x < sizeof(_buffer); x++)
  {
    _buffer[x] = data;

#if defined(ESP8266)
    if (x % 1000 == 0) {
      yield();
      Serial.println(ESP.getFreeHeap());
    }
#endif
  }

  Serial.print("fillScreen: leave");
}
```

Console output directly after flashing:

```text
fillScreen enter
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Console output after power disconnect (reboot):

```text
fillScreen enter
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v794630e0
~ld
```

Things I also tried, but with no real change to the issue:

```c++
void GxGDEW075T8::fillScreen(uint16_t color)
{
  uint8_t data = (color == GxEPD_BLACK) ? 0xFF : 0x00;
  std::uninitialized_fill_n(_buffer, GxGDEW075T8_BUFFER_SIZE, data);
}
```

```c++
void GxGDE0213B1::fillScreen(uint16_t color)
{
  uint8_t data = (color == GxEPD_BLACK) ? 0xFF : 0x00;
  for (uint32_t x = 0; x < sizeof(_buffer); x++)
  {
    _buffer[x] = data;
  }
}
```

```c++
void GxGDE0213B1::fillScreen(uint16_t color)
{
  uint8_t data = (color == GxEPD_BLACK) ? 0xFF : 0x00;
  for (uint32_t x = 0; x < sizeof(_buffer); x++)
  {
    _buffer[x] = data;

#if defined(ESP8266)
    if (x % 1000 == 0) delay(100);
#endif
  }
}
```

Just to make sure, that it is library/ESP8266/resources issue, I tried to not fill the complete buffer (only first 10.000 entries). And it worked! (I can see the display "flashing" / updating)

```c++
void GxGDE0213B1::fillScreen(uint16_t color)
{
  uint8_t data = (color == GxEPD_BLACK) ? 0xFF : 0x00;
  for (uint32_t x = 0; x < 10000; x++)
  {
    _buffer[x] = data;
    
#if defined(ESP8266)
    if (x % 1000 == 0) {
      yield();
      Serial.println(ESP.getFreeHeap());
    }
#endif
  }
}
```

```text
fillScreen enter
48328
48328
48328
48328
48328
48328
48328
48328
48328
48328
fillScreen: leave
```

As I said, I'm open for any help. I don't know how to continue fixing this "bug".

G6EJD, 2018-10-01:
Your out of memory on the ESP8266. What size buffers are you setting for the JSON object. The driver builds and image in a buffer (large memory requirement) and how much JSON data are you trying to decode.

SunboX, 2018-10-02:
Hm, are you sure? Because `ESP.getFreeHeap()` tells constantly 48328. Can you explain me this behavior? Is it not up to date with the real memory available?

SunboX, 2018-10-02:
If I get the time, I post a small sample sketch showing the issue today in the evening.

ZinggJM, 2018-10-02:
The 7.5" b/w e-paper display consumes almost all RAM on ESP8266 for graphics buffer. Take a look at the compiler output in the Arduino IDE. Your heap values can't be correct. Your Jason may hang in the background, most likely.
I propose you use GxEPD2 for your application. There you can control the graphics buffer size by the template parameter page_height.

SunboX, 2018-10-02:
Hi, @ZinggJM ! 😄 

> The 7.5" b/w e-paper display consumes almost all RAM on ESP8266 for graphics buffer. 

Ok, good to know. 😞 It's confusing that it reports such free memory. For a moment I thought it might be out of RAM, that's why I put the debug output there. But the free memory reporting told me, that this is not the issue... 😢 

> Take a look at the compiler output in the Arduino IDE. Your heap values can't be correct.

Compiler output also looked ok to me:
```text
Build-Optionen wurden verändert, alles wird neu kompiliert
Archiving built core (caching) in: /var/folders/6d/h53dfksj169fwbfs0nqgx7s80000gn/T/arduino_cache_326488/core/core_esp8266com_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M3M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_05bb99f249761b9b37b8de2c0d5f45c2.a
Der Sketch verwendet 343304 Bytes (32%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 31456 Bytes (38%) des dynamischen Speichers, 50464 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
warning: serialport_set_baudrate: baud rate 921600 may not work
Uploading 347456 bytes from /var/folders/6d/h53dfksj169fwbfs0nqgx7s80000gn/T/arduino_build_311603/OpenSenseDisplay.ino.bin to flash at 0x00000000
................................................................................ [ 23% ]
................................................................................ [ 47% ]
................................................................................ [ 70% ]
................................................................................ [ 94% ]
....................                                                             [ 100% ]

```

> our Jason may hang in the background, most likely.

Don't think so. It should be thrown away before the display init. But maybe it's a combination of WifiClientSecure (that need a lot of memory) and the JSON lib. If everyone does not free all RAM internally, it might be the case that it's running out of it.

> I propose you use GxEPD2 for your application. There you can control the graphics buffer size by the template parameter page_height.

Oh, great. I didn't see this before. I really really appreciate your work! ❤️ This one works like a charm! Thank your very much!

/closed

closed 2018-10-02 by SunboX