ZinggJM/GxEPD #57: Could you please add `git tags` for each version?

roschaefer, 2018-10-14:
I had this [trouble here](https://github.com/sqfmi/badgy/pull/25) and openend a PR for it. I am used to have `git tags` for each released version but this repository has none. So I had to guess what's the latest v2.X release.

ZinggJM, 2018-10-15:
This is an enhancement request, not an issue with the library or its use with Arduino.
The form of this request is not suited to motivate me to do anything.

closed 2018-10-15 by ZinggJM

roschaefer, 2018-10-15:
Why is the "form if this request not suited" to motivate you to do "anything"? If I did something wrong I would be glad if you could explain it to me.

ZinggJM, 2018-10-15:
I don't like issues that are not real issues. I had worked for a big German company, where "quality reports" were misused for enhancement requests, just to account costs to a different budget.

I expect at least some explanations for enhancement requests, not just a link to unclear information.
There may be good reasons for wanting version tags, e.g. for reference from other sources put on GitHub, but you could take the effort to explain your need.

roschaefer, 2018-10-15:
Here you go:
I wanted to run the examples of the repository of the referenced PR. In order to that, I need to install four dependencies. This library happens to be one of them. When I install this library I get compile errors. I suspect a version constraint because the README tells me to use `v2.X` of this library. I see in the README of this library, that the latest version seems to be `v3.X`. I want to check out the latest `v2.X` release. But I don't know which commit that is. The `git log` doesn't help, because the commit messages start to indicate the version starting at `v3`, not before. There are no releases or git tags on Github either. 

So I think, I should tell the maintainer to add tags indicating the released version. I cannot do it myself, because I need write permission to the repository. So I open an Issue. The issue references a PR for context, you can see in the changed files what information I need.

So, the solution to all of this is to go through this list [here](https://github.com/ZinggJM/GxEPD/commits/master/library.properties) and tag all commits with the respective version. This is a small but helpful improvement of this repository. Closing the issue wouldn't hurt me in particular, because I solved my issue already.

What is more interesting is the refusing reply of the maintainer, so I ask myself, did I do something wrong? I actually wanted to help the maintainer to make the repository better.