ZinggJM/GxEPD #58: 7,5" E-Paper with ESP32 - do not update image

SazZzka, 2018-11-30:
Hi everyone!

Asking you for help.

I have bought  a 7.5inch Waveshare e-Paper HAT (C) 
and 
E-Paper ESP8266 Driver Board (https://www.waveshare.com/wiki/E-Paper_ESP8266_Driver_Board) with ESP-12F chip (if this information will help)

The main Idea - to display a static image and text on my E-Paper display.
When I open Demo Code (https://www.waveshare.com/wiki/File:Loader.7z) compile and run project - everything is work, image update.

However, when Im trying using examples from GxEPD library - GxEPD_Example or GxFont_GFX_Example nothing happening with the display.
I do not get any errors, project compile, blue LED blinking.

Display class - #include <GxGDEW075Z09/GxGDEW075Z09.h>    // 7.5" b/w/r

In the IDE I  select NodeMCU 0.9 (ESP-12 Module) 

But the display simply does not react. 

Please help me - what I am doing wrong?

G6EJD, 2018-11-30:
SPI WiFi g mismatch between what the waveshare ESP8266 board is using and what pins are being used, check the waveshare wiring configuration and make sure the EPD library pins defined are the same.

SazZzka, 2018-11-30:
Sorry for asking such a silly question but where exactly I should check and compare it?
As far as I understand I should modify some information here?
-------------------------------------------------------------------------------------------------
#if defined(ESP8266)

// for SPI pin definitions see e.g.:
// C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\generic\common.h

GxIO_Class io(SPI, /*CS=D8*/ SS, /*DC=D3*/ 0, /*RST=D4*/ 2); // arbitrary selection of D3(=0), D4(=2), selected for default of GxEPD_Class
GxEPD_Class display(io /*RST=D4*/ /*BUSY=D2*/); // default selection of D4(=2), D2(=4)
// Heltec E-Paper 1.54" b/w without RST, BUSY
//GxEPD_Class display(io, /*RST=D4*/ -1, /*BUSY=D2*/ -1); // no RST, no BUSY
-----------------------------------------------------------------------------------------------

Where to get - waveshare wiring configuration?

Iguhs, 2018-11-30:
Hi,

the configuration for E-Paper ESP8266 Driver Board :
         GxIO_Class io(SPI, 15, 4, 5);
         GxEPD_Class display(io, 5, 16);

SazZzka, 2018-11-30:
> Hi,
> 
> the configuration for waveshare E-Paper ESP8266 Driver Board :
> GxIO_Class io(SPI, 15, 4, 5);
> GxEPD_Class display(io, 5, 16);

Hi, ok I will try in a few hours. Please tell me where did you get this information?

Iguhs, 2018-11-30:
Yes, in the document "[ESP8266_e-paper.pd](https://www.waveshare.com/w/upload/7/7e/ESP8266_e-paper.pdf)" you can see.

G6EJD, 2018-11-30:
Have you checked the ribbon cable connection, maybe disconnect and reconnect it. 
Is there any display activity at all?
It could be a faulty display/ driver.

SazZzka, 2018-11-30:
> Have you checked the ribbon cable connection, maybe disconnect and reconnect it.
> Is there any display activity at all?
> It could be a faulty display/ driver.

Yes, when I use Loader again, everything work, images uploades.
[image]

Moreover, what I have found, this "strip" still left print on the display.

I have no idea whats happenning

G6EJD, 2018-11-30:
Contact Waveshare send them a picture it looks like it’s faulty to me.

Try an all black screen band still there?
Try an all white screen band still there?
And the same test for yellow?

Iguhs, 2018-11-30:
My displays need several loads and unloads of differents images to forgive the shadows of the factory default image.

SazZzka, 2018-11-30:
For the shadows I dont care now. The main is - why do  GxEPD_Example not work :( I just checked another sample from GxEPD library (Partial Update Library) and seems to me it is work. Im not sure correct or not but it show image 
GxEPD_Example still not

ZinggJM, 2018-12-01:
If you follow the issue_template.md you get a better chance to get an answer from me!

**And it is not a big effort to read at least through the README.MD**

#### Version 2.3.9
- new version for 7.5" 3-color display GxGDEW075Z09
- GxGDEW075Z09 runs with full buffer on ESP32, Arduino Due, STM32F4
- runs with reduced buffer on ESP8266, STM32F1, AVR : will show buffer content as stripes
- supports paged display on AVR, ESP8266, STM32F1

> **will show buffer content as stripes**, 2 parts in case of ESP8266.

I propose you try the GxEPD2_Example of https://github.com/ZinggJM/GxEPD2, it is easier to use with paged drawing.

ZinggJM, 2018-12-06:
Closed because got no more input, no response.

closed 2018-12-06 by ZinggJM