ZinggJM/GxEPD #59: Accepting donation devices?

Norfeldt, 2018-12-01:
> **I can and will only support e-paper panels I have!**
> * promotion panels from suppliers are welcome, to add support to GxEPD
> * donation panels from users are welcome, to add support to GxEPD

Would you accept a donation of a Particle Photon?
[image]

Would like to make a port of GDEW075Z09 7.5" b/w/r to be working with GxEPD on the Photon.
https://community.particle.io/t/waveshare-7-5-epaper-b-w-r-and-gxepd-porting/45826

Would like to show some images that I'll store online (and preprocess) and have it print them.

ZinggJM, 2018-12-02:
Yes, I would be pleased to get that donation, and I would like to try to get GxEPD and GxEPD2 work on that board.
Can this board be used with the Arduino IDE (is there a package for it)? Does it have HW SPI?
You could help me get started with some more links.
You could also consider asking if someone has already used it with your board in the Arduino Forum Display. In the Arduino Forum I can send my address in a personal message; I don't know how to do this in GitHub. Thanks for your question.

ZinggJM, 2018-12-02:
It doesn't look like there is a package for the Arduino IDE for this board.
The Particle Photon looks interesting enough for me to buy to learn about and its development environment. But when I came to the page adding shipping cost, I decided against buying it. More than twice the cost of the board ($19) does not compare well to the boards I am used to, mainly ESP8266, ESP32 and STM32F1. And porting to a different IDE would be a low priority task anyway.

per1234, 2018-12-02:
It is possible to make Arduino libraries also compatible with the Particle IDE. [Particle uses the standard Arduino API](https://docs.particle.io/tutorials/device-os/libraries/#overview) for the most part.

Unfortunately, I often see the Particle community take working Arduino libraries and then intentionally break their Arduino IDE compatibility in the process of "porting" to Particle. Then they continue to advertise them as an "Arduino" library, meaning I end up with that as garbage in my search results. It's really a shame that the Particle organization and community doesn't operate in a manner more complementary to the Arduino project. That would benefit both communities.

ZinggJM, 2018-12-02:
Thank you @per1234 for this information. My priority is for Arduino, so I will be cautious.
@Norfeldt so I assume I could make the library compatible, most likely GxEPD2, if you still like to donate.

Norfeldt, 2018-12-03:
> @Norfeldt so I assume I could make the library compatible, most likely GxEPD2, if you still like to donate.

Yes sir, I stand by my offer. I have PM'ed you on the Arduino display forum.

ZinggJM, 2018-12-04:
@Norfeldt 
Thank you sir, I have PM'ed you my address. I am looking forward to supporting Particle Photon.
If you like to get me started in advance, you could provide a link to download the particle development environment, and maybe a tutorial. I assume the link above gets me to the particle forum. Thank you!

Norfeldt, 2018-12-04:
Sounds great 👍 

https://www.particle.io/developer-tools

There is an offline IDE, but I have always been using the online one that is located at
https://build.particle.io

It has some decent docs
https://docs.particle.io

This might be relevant
https://docs.particle.io/tutorials/developer-tools/build/photon/#contribute-a-library

Norfeldt, 2018-12-04:
Don't know if you will find this "hello world" tutorial interesting

https://www.youtube.com/watch?v=b6sUP16HWKM

Norfeldt, 2018-12-04:
Try this:
https://docs.particle.io/tutorials/developer-tools/dev/

ZinggJM, 2018-12-04:
> All the files have to be on the same level (no subdirectories) like this

I just wonder how library developers cope with this. Some sort of a hen/egg problem?
Does this mean libraries also have to be flat (all files in one directory)?

Norfeldt, 2018-12-04:
Another example I saw working for some 2 inch display had a flat folder structure - I don't think is pretty or structured.

I saw one guy on the forum mentioning that it would be possible to do some folder structure in the near future.

pkourany, 2018-12-04:
You could use the [Particle CLI](https://docs.particle.io/tutorials/developer-tools/cli/#installing) which lets you use a local folder structure for libraries and code but compiles in the Cloud.

Norfeldt, 2018-12-06:
Should arrive at you soon (most likely today). 

The status is "With delivery courier".

ZinggJM, 2018-12-06:
It has arrived, thank you. I close this issue, as it is not an issue with this library, and will hopefully be solved in the Particle eco-system. You still can add comments. I will add a link to the Particle forum, as soon as I report or ask for more help.

closed 2018-12-06 by ZinggJM

Norfeldt, 2018-12-06:
Sounds great! Merry Christmas 🎄🎁🎅

ZinggJM, 2018-12-10:
I am struggling with the Photon development tools. Particle Desktop IDE doesn't work well with libraries, and Adafruit_GFX for Particle is limited, and Adafruit_GFX_RK doesn't compile (for me).
But I have created the repository PxEPD2, where I put preliminary steps as Particle project under examples.

Norfeldt, 2018-12-13:
Wow, that was fast! Thank you so much!!

 I got the demo up and running. Now I just need to figure out how to drop some photos to it. I notice that 
 
```
// not enough code space
```

in 
```
// select only one to fit in code space
#include "bitmaps/Bitmaps200x200.h" // 1.54" b/w
//#include "bitmaps/Bitmaps128x250.h" // 2.13" b/w
//#include "bitmaps/Bitmaps128x296.h" // 2.9"  b/w
//#include "bitmaps/Bitmaps176x264.h" // 2.7"  b/w
////#include "bitmaps/Bitmaps400x300.h" // 4.2"  b/w // not enough code space
////#include "bitmaps/Bitmaps640x384.h" // 7.5"  b/w // not enough code space
// 3-color
//#include "bitmaps/Bitmaps3c200x200.h" // 1.54" b/w/r
//#include "bitmaps/Bitmaps3c104x212.h" // 2.13" b/w/r
//#include "bitmaps/Bitmaps3c128x296.h" // 2.9"  b/w/r
//#include "bitmaps/Bitmaps3c176x264.h" // 2.7"  b/w/r
////#include "bitmaps/Bitmaps3c400x300.h" // 4.2"  b/w/r // not enough code space
```

So I would need to chop the photo into smaller fractions and print each one at a time - if possible?

ZinggJM, 2018-12-14:
Hi Norfeldt,

You already found the first error!

This was the part for AVR, not the general one, of the GxEPD2_Example. it should be:

```// comment out unused bitmaps to reduce code space used
#include "bitmaps/Bitmaps200x200.h" // 1.54" b/w
#include "bitmaps/Bitmaps128x250.h" // 2.13" b/w
#include "bitmaps/Bitmaps128x296.h" // 2.9"  b/w
#include "bitmaps/Bitmaps176x264.h" // 2.7"  b/w
#include "bitmaps/Bitmaps400x300.h" // 4.2"  b/w
#include "bitmaps/Bitmaps640x384.h" // 7.5"  b/w
// 3-color
#include "bitmaps/Bitmaps3c200x200.h" // 1.54" b/w/r
#include "bitmaps/Bitmaps3c104x212.h" // 2.13" b/w/r
#include "bitmaps/Bitmaps3c128x296.h" // 2.9"  b/w/r
#include "bitmaps/Bitmaps3c176x264.h" // 2.7"  b/w/r
#include "bitmaps/Bitmaps3c400x300.h" // 4.2"  b/w/r
```
In fact, I only did very rudimentary tests so far, as I was struggling with the Particle tools.

See https://community.particle.io/t/adding-libraries-to-project-local-and-public-ones/46056

Norfeldt, 2018-12-14:
@ZinggJM I'm not chasing errors and I can see that you have done a tremendous work! Wow 🥇 

I just want to learn how to draw some bitmap images that I put in an 'images.h' file

So far I have added this 

```
#if defined(ESP8266) || defined(ESP32)
#include <pgmspace.h>
#else
#include <avr/pgmspace.h>
#endif

const unsigned char gImage_youTubeImage[11400] PROGMEM= { /* 0X00,0X01,0X2C,0X01,0X2C,0X01, */
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
....
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,};
```

and written the following 

```
#include <GxEPD2_PP.h>
#include <Adafruit_GFX_RK.h>
#define ENABLE_GxEPD2_GFX 0
#include <Arduino.h>
#include <Adafruit_GFX.h>
#include <FreeMonoBold9pt7b.h>
#include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>

#if defined(PARTICLE)
GxEPD2_3C<GxEPD2_750c, GxEPD2_750c::HEIGHT / 4 > display(GxEPD2_750c(/*CS=D5*/ SS, /*DC=*/ A1, /*RST=*/ A0, /*BUSY=*/ D4));
#endif

#include "images.h"

void setup()
{
  delay(1000);
  display.init(115200);
  display.drawImage(gImage_youTubeImage, 0, 0, 300, 300, false, false, true);
}

void loop()
{
}
```

But it just throws this error at me:

```
call of overloaded 'drawImage(const unsigned char [11400], int, int, int, int, int, int, int)' is ambiguous
```

G6EJD, 2018-12-15:
Your image is a not an integer result, so 300 wide / 8 then x 300 deep = 11250 bytes that first conversion to bytes = 37.5, try changing that to an integer result. Why are you using 11400?

Perhaps try a small image first so you can see what the image format is.

ZinggJM, 2018-12-15:
I will take a look at your issue. But I would prefer to continue this discussion in a forum.

I have created a topic in the Particle Forum for this:

https://community.particle.io/t/gxepd2-pp-particle-display-library-for-spi-e-paper-displays/46305

Norfeldt, 2018-12-15:
@G6EJD I have tried to answer your question on the Particle Forum thread that ZinggJM created.

https://community.particle.io/t/gxepd2-pp-particle-display-library-for-spi-e-paper-displays/46305/2?u=norfeldt