ZinggJM/GxEPD #60: Display lock after drawBitamp() on GDEW0213I5F 2.13" b/w 104x212 flexible

habkorn, 2018-12-05:
Hi,

I really like that you added support for the flexible GDEW0213I5F 2.13" b/w 104x212.

However, using the example "GxEPD_Example"  this particular display locks after successfully drawing one bitmap.

Again, with more details:
In the line in "GxEPD_Example"
display.drawExampleBitmap(BitmapExample1, sizeof(BitmapExample1));

"BitmapExample1" is drawn. Serial montitor output is also ok, so the program is not terminated. However, the display does not show the second Bitmap commanded by the line 
display.drawExampleBitmap(BitmapExample2, sizeof(BitmapExample2));

Do you know of a fix for this problem?

By the way: the same code works fine on other displays, like GDEP015OC1.h 1.54" b/w

Kind regards, keep up the good work!

ZinggJM, 2018-12-06:
Did you notice the issue_template.md? You did not tell what board/processor you use.

I have just repeated the test with ESP8266 Wemos D1 mini: works fine.
With 3.3V Arduino Pro Mini I get erratic behavior; sometimes it works, sometimes it does reset loops, sometimes it hangs. I don't know why I did not notice this on my tests.
So this needs more tests.

ZinggJM, 2018-12-06:
Test with Arduino UNO with voltage divider proto shield for data lines works fine!
The LDO on the USB to Serial converter board I use with the 3.3V Arduino Pro Mini seems too weak for this display.

So I need the additional information about this issue, else I will close it.

habkorn, 2018-12-09:
Hi, thanks for your info!

I'm using a "MH-ET LIVE MiniKit for ESP32" , see http://mh.nodebb.com/topic/8/mh-et-live-minikit-for-esp32

To use this board, GxIO and GxEPD are configured like this:

GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 16, 4);

This works well using GDEP015OC1, but not for GDEW0213I5F. In the latter case, the display shows only the first image (at least for me).

Minimum example:
[min_ex.ino.txt](https://github.com/ZinggJM/GxEPD/files/2660784/min_ex.ino.txt)

From the tests you did with ESP8266 Wemos D1 mini, maybe you can derive what's wrong with the MH-ET board. This board has Bluetooth and lower deep sleep current than ESP8266 Wemos D1 mini, would be nice if it's a problem easy to fix.

Kind regards,
Thomas

habkorn, 2018-12-11:
Hi, 
thank you for the tip with the ESP8266 Wemos D1 mini, it works right out of the box with the flexible display and your library.

For the "MH-ET LIVE MiniKit for ESP32", which only is able to show the first image, the log output is
[image]

I've tested 3 seperate MH-ET Live boards, all show the same behavior with the same display which did work with ESP8266. I also tested a Sparkfun ESP32 Thing -- same behavior.

I find it rather strange that this problem only affects the flexible display, but not GDEP015OC1 1.54" b/w. 

Possible resolution:

I said before, I use for initalization (because the default in GxEPD_Example produces no display output)
GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 16, 4);

.. and connect display and board in the following way:

BUSY -- IO4, RST-- IO21, DC -- IO22, CS -- IO5,  CLK -- IO18, DIN -- IO23, GND -- GND, 3.3V -- 3.3V

Meanwhile, I think I did connect a pin or two wrong or initialized the GxIO_Class or GxEPD_Class wrong in the first place. This didn't affect GDEP015OC1, but the flexible display is more picky I guess.

Searching on the web for the correct connections did not yield a working result. Please advise if you have an idea.
My workaround in the meantime is to use ESP8266 Wemos D1 Mini and get on with my life.

Kind regards,
Thomas

G6EJD, 2018-12-11:
BUSY -- IO4, RST-- IO21, DC -- IO22, CS -- IO5, CLK -- IO18, DIN -- IO23, GND -- GND, 3.3V -- 3.3V
Contains a typo - I think, it should be CLK -- IO16 - checking
One technique worth trying, is to switch on the pullups on the ESP32 gpio pins in-use, this may assist with the display driver device as its input characteristics seem to be different from other display types.
pinMode(4, INPUT_PULLUP); and then the same for 21, 22, 5, 16, 23, I checked all those pins support pull-ups.

ZinggJM, 2018-12-11:
This may be an issue with BUSY becoming active too late. I think I have seen a delay in the Good Display demo code, but it worked without it in my tests. You could add delay(2); in _waitWhileBusy() before entering the while loop, for a test. Please report if that helps.

habkorn, 2018-12-12:
Hi, thanks for the quick reply.
when I connect CLK to IO16, no output is shown.
For CLK -- IO18, at least the first image is shown. 

Might me some mislabeling on the board? I can't say.

I also tried

pinMode(4, INPUT_PULLUP);
pinMode(21, INPUT_PULLUP);
pinMode(22, INPUT_PULLUP);
pinMode(5, INPUT_PULLUP);
pinMode(18, INPUT_PULLUP);
pinMode(23, INPUT_PULLUP);

..and inserted delay(2);

[image]

However, behavior did not get better.

I'm open to try out some other fix if you have one, and also send you a MH-ET board if you like.
For now, I'll stick with the Wemos board.

Thanks for your effort!

G6EJD, 2018-12-12:
Is your programme using serial communications? if so IO16 an 17 are used for serial services and there will be a conflict.

If you connect another type of display to the same pins then does that operate? I think yes, in-which case there is a timing or I/O issue. 

It could be an SPI bus speed issue, maybe it needs to be slowed down, but I haven’t checked the data sheet to suggest a speed value, if that’s the issue.

ZinggJM, 2018-12-12:
Always observe diagnostic output when something doesn't work.
If BUSY time is short as you reported (<10000us), either SPI is not working, or BUSY line is not working.
If BUSY is too short, but you see output on the display, then BUSY is not working.
You could post a photo of your wiring.

G6EJD, 2018-12-12:
I'm still confused by the wiring, I think it's wrong:
GxIO_Class io(SPI, CS, DC, RST);
GxEPD_Class display(io, RST, BUSY);

You have:
GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 16, 4);

So in one instance you have RST = 21 and the other RST = 16

The correct definition should be:

GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 21, 4);

?

ZinggJM, 2018-12-12:
Aha. This explains the behaviour. If RST is correct for io, but not for display, then the display will once be woken correctly by io, but never again after any deep sleep powerDown. Use of deep sleep is class dependent, not consistent throughout GxEPD (leftover from demo code).

G6EJD, 2018-12-14:
Hey @habkorn did this solution work for you?
It would be nice to say either way, as we spent some time trying to help...

ZinggJM, 2018-12-14:
Timeout! Closed.

closed 2018-12-14 by ZinggJM

habkorn, 2018-12-15:
Hi! Sorry about my late reply.

Your time was effective, because

GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 21, 4);

WORKED!!

The wiring on MH-ET Live Esp32 Minikit is

BUSY -- IO4, RST-- IO21, DC -- IO22, CS -- IO5, CLK -- IO18, DIN -- IO23, GND -- GND, 3.3V -- 3.3V

Your effort is much appreciated!

habkorn, 2018-12-23:
> Timeout! Closed.

Hi! Sorry about my late reply.

Your time was effective, because

GxIO_Class io(SPI, SS, 22, 21);
GxEPD_Class display(io, 21, 4);

WORKED!!
I can see now where my error was.
The wiring on MH-ET Live Esp32 Minikit is

BUSY -- IO4, RST-- IO21, DC -- IO22, CS -- IO5, CLK -- IO18, DIN -- IO23, GND -- GND, 3.3V -- 3.3V

Your effort is much appreciated!

Kind of interesting that my initial wrong definition worked on GDEP015OC1..

Kind regards

G6EJD, 2018-12-23:
Some display controllers need a reset some not, so probably down to the controller being different.
Good to hear it’s working.