ZinggJM/GxEPD #63: 2.9" bwr How to controll border

atc1441, 2019-01-13:
Hi, i realy like your Library, 

i have a feew 2.9" Display's that are BW but only work in the 2.9" BWR file.

on my display the black is in the library handled as red, so the colors are switched but that isn't my problem.

it always shows an black border around the 128x296 pixels how could i turn this border on or off? 

thanks

ZinggJM, 2019-01-14:
It is a always good idea to post a link to the shop you bought the display from. It might be an interesting shop for the display with case. The panel might well be from Good Display. The sticker would be hidden on the backside of the panel. The inking on the flex connection might be interesting, is it WFT029CZ10 LW? I will first update GxEPD2 with support for GDEW029T5, then GxEPD.

Did you buy these as shelf labels, or do you produce shelf labels?

What is your native language? Your displays seem to talk German.

atc1441, 2019-01-14:
Unfortunately i bought these used on eBay, so there is no shop, i paid around 2$ per display. they were used in a shop around 2016 and have a Samsung  SEM9110 MCU inside with zigbee connection but i remove the Samsung chip and connect to the spi bus directly because there is no documentation for the chip. 
 
German is my native language, i think i wrote it wrong in the post before. 

i add a picture from the backside of one Display on the flex connection is WFS0290CF19 LW printed.

in this PDF every command is very well explained, i would bet you know it :D http://www.e-paper-display.com/download_detail/downloadsId=535.html

As in the description for Command 0x50 the first two bits are for the border if i set them i can turn the border on or off with the display being inverted.

now i added a function that is called "setBorder(bool);" this works as expected:

```
 _writeCommand(0X50);   
if (_border)_writeData(0x77); else _writeData(0xB7); 0b01110111 or 0b10110111
```

[image]

closed 2019-01-14 by atc1441