ZinggJM/GxEPD #64: impossible use with arduino DUE

claudius310, 2019-01-14:
hello,
I used the example with Arduino MEGA 2560 the screen works perfectly
I used Arduino DUE with the SPI setting:
GxIO_Class io(SPI, /*CS=*/ 21, /*DC=*/ 52, /*RST=*/ 53); 
GxEPD_Class display(io, /*RST=*/ 53, /*BUSY=*/ 51);  
and  I connected CLK -> 76, DIN -> 75
but it does not work
Thank you for your help

ZinggJM, 2019-01-14:
**you totally ignored the issue template**

It **is** possible to use with Arduino Due.

closed 2019-01-14 by ZinggJM

claudius310, 2019-01-14:
Hello and thank you for your answer,
I do not understand the first sentence well, I am in France and I speak English badly.
if it is possible to use Arduino Due, how to connect it and how to use your library in the simple example?
regards

claudius310, 2019-01-14:
hello,
OK I had confused the number i / o CS.
Now the example works with Arduino Due
thank you for your library

Best regards
Jean-Claude