ZinggJM/GxEPD #66: where is GxGDE0213B72.h ?

coolty, 2019-04-06:
Where is GxGDE0213B72.h? Many projects reference this specific repo and mention this file, but it's not in this repo. Where did it go?

G6EJD, 2019-04-06:
That version is specifically from: https://github.com/lewisxhe/TTGO-Electronic-Badge
the prolithic copier of others code often without recognition of the author.

coolty, 2019-04-06:
Well it's not on that repo either and it tells us to use your repo for
those files.. were they removed from the repo?

ZinggJM, 2019-04-06:
From README.md:
> ### I can and will only support e-paper panels I have!
- promotion panels from suppliers are welcome, to add support to GxEPD
- donation panels from users are welcome, to add support to GxEPD

I don't have that display, and the file mentioned was never part of my library.
I should get that e-paper soon from Good Display, as the original 2.31" b/w is no longer available.
I will then ad **my** version to my libraries.

@coolty 
You should do some search and background check before opening such an issue!

**There are so many forks of this library; I can't have an overview of who added what to his version.**

closed 2019-04-06 by ZinggJM