ZinggJM/GxEPD #71: Problems with 7.5 inch rbw display and Waveshare ESP32

jofleck, 2019-05-30:
Hi since a few days I try to get my 7.5 inch rbw displays working with this library. I designed a PCB with an ESP8266 (I used your recommended pinout). With this library the display does exactly nothing but smaller displays are working like a charm. I thought it was my fault (something wrong with the PCB) but then I tried another working ESP8266 board and saw the same behavior. With Waveshares EPD example sketch everything works fine. 

I decided to buy Waveshares original ESP32 driver board and noticed that it isn't possible to change the Display SPI pins in your library. Is there any solution to get the original ESP32 board working with your library?

Thank you very much :)

ZinggJM, 2019-05-30:
You're welcome!

But you might have had a look into the Arduino Forum Displays topics concerning my libraries, mentioned in the README.MD.
Your first issue mentioned isn't clear to me, it might be you need a different pin for CS; SS may interfere with ESP8266 boot modes. Take a look at diagnostic output in Serial Monitor.

For ESP32 driver board take a look at the example for it in GxEPD2:
https://github.com/ZinggJM/GxEPD2/tree/master/examples/GxEPD2_WS_ESP32_Driver
I have not tried, but the same method should also work with GxEPD.

Good Luck!

G6EJD, 2019-05-30:
Depending on the programme your trying to use with an ESP8266 and 7.5” generally there is I sufficient memory to form the image, 4.2” is usually OK.
There are plenty of examples including one in the EPD library for using the Waveshare driver board. Don’t expect low power consumption with that board though, only the Lolin D32 can achieve uA levels at sleep. 
The EPD libraries allow all the lines to be defined, but if you want to remap the SPI bus you’d need to use SPI.begin(MOSI, CLK, etc it’s in the documentation

ZinggJM, 2019-05-31:
@jofleck 
Please provide more details about your issue(s), if you need more help, or close this issue.

closed 2019-06-01 by ZinggJM

jofleck, 2019-06-01:
Thanks☺️ For the moment everything is working ☺️