ZinggJM/GxEPD #72: SD Examples fail to mount

9H5G, 2019-05-30:
## Please post only issues that are real issues with this library!
- please use the Arduino Forum Displays for questions or enhancement suggestions

### Version number of the library used, should be the current one
- this number is needed for later reference and tracking
3.0.7
### Expected Behavior
Expected that the SD card would mount
### Actual Behavior
SDcard fails to mount
### E-Paper module used
T5 V2.3 2.13 (b&W)
### Arduino / Processor module used
ESP32
### Steps to reproduce the behavior or reduced but complete code example
Change line 43 to define SD_CS 13
Change line 44 to define EPD_CS 05
Edit #include for screen type

Compile and run.  Message "SD Failed" generated at #172

I suspect I misconfigured the pins but there is insufficient documentation to set the correct pins (MOSI - 15, SCK - 14, MISO - 2, CS - 13)

ZinggJM, 2019-05-30:
This is not an issue with this library. If you successfully checked with a standalone SD example, and then can't get the SD example of this library to work, you then can re-open this issue.

closed 2019-05-30 by ZinggJM

G6EJD, 2019-05-30:
The way to solve this is determine which pins the T5 is using for its SPI bus, which can be difficult, but from my version this is what I used:
MOSI: 23 (SPI MOSI)
MISO: 19 (SPI MISO)
SCK: 18 (SPI SCK)
SS: 5 (the display CS)
DC: 17 (SPI DC)
RST: 16 (the display RESET)
BUSY: 4 (the display BUSY)

For the SD-Card you'd need a CS pin so choose 13 which should be OK, providing you initialised as SD.begin(13);

I would enable pull-ups on the MOSI and CS lines as I believe they are missing on the board.

These boards are that good...

G6EJD, 2019-05-30:
Has the board been redesigned? I found this in the library under T5:

#define SCK_Pin	18
#define MOSI_Pin	23
#define MISO_Pin	19
#define DC_Pin	26
#define BUSY_Pin	32
#define RST_Pin      27
#define CS_Pin		5

Try those.

9H5G, 2019-05-30:
Thanks Dave, It seems there has been a reassignment. (See photo) [https://photos.app.goo.gl/q4HPKrhp6pYwp7em6](url)
Also documented at: https://github.com/lewisxhe/TTGO-Electronic-Badge

The library is very well documented apart from this aspect but I note the author's ascerbic comments and will crawl off back into my cave.

G6EJD, 2019-05-30:
OK, so 2 SPI buses have been implemented that’s unusual, the board has moved on since I got mine. So there must be an SPI2.begin(pin,pin, etc) in their examples? Email me for further discussion and assistance if I can.

ZinggJM, 2019-05-31:
@9H5G 
Do you really believe this is an issue with **my** library?
Or isn't it rather an issue with one of the forked libraries:
https://github.com/LilyGO/GxEPD
https://github.com/lewisxhe/GxEPD

I don't understand why I should be obliged to document TTGO specifics.

Please post the issue at the appropriate place.

@G6EJD 
I made mostly negative experience with TTGO hardware and software.
And now they are getting on my nerves with their use of my library.

G6EJD, 2019-05-31:
I agree with you, they (TTGO) undertake blatant copyright infringement, I’ve successfully obtained Take Down (GitHub speak for removal) of material by them, you could do the same. This is why the US Government are launching the drive against improper Chinese use of others information and materials. I do have empathy with you and have been equally frustrated by them.

9H5G, 2019-05-31:
I was only saying that your example fails. Thank you for such a ROBUST
response.

If you don't want comments on examples, please remove them or at least make
a note that you do not support the use of your libraries (or the examples)
on specific hardware. I cannot be expected to know your mind or know the
ins and outs of every hardware manufacturer and be abused for making the
report I have.

ZinggJM, 2019-05-31:
I wish you Good Luck with your TTGO hardware.